adafruit/Adafruit_nRF52_Arduino #24: nrfutil python 3 setup issue on OSX/Linux

hathach, 2017-03-27:
https://forums.adafruit.com/viewtopic.php?f=57&t=114274

microbuilder, 2017-04-05:
I added an FAQ on this issue until OS X binaries are available: https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/arduino-bsp-setup#faq-21

hathach, 2017-04-05:
It happens with python 2 on osx as well. The user has responded

macOS = 10.12.3 (current release is now 10.12.4)Python = 2.7.10

hathach, 2017-04-19:
OSX binary failed with python 2.7.10, but work with version 2.7.13

hathach, 2017-07-13:
maybe some days we could try to implement this nrfutil using C ?

microbuilder, 2017-07-13:
I think C might actually be the best approach in the future, yes. The problem here is reimplementing the protocol from Nordic, but if we do it once in C it will cover all the platforms and there are some previous examples we can recycle for some of the cross platform code.

hathach, 2018-08-09:
done, thanks to @dhalbert

closed 2018-08-09 by hathach