adafruit/Adafruit_nRF52_Arduino #28: Windows 10 bsp installation issue

hathach, 2017-03-30:


hathach, 2017-04-07:
couldn't reproduce on my windows 10 laptop !!! User may need to use the manual method !!!!

microbuilder, 2017-04-07:
Is it possibly related to this being installed on Windows 10 machines? https://www.howtogeek.com/249966/how-to-install-and-use-the-linux-bash-shell-on-windows-10/

hathach, 2017-04-07:
I install this ubuntu bash already, along with tons of other linux stuffs like cygwin, mingw msys etc ... maybe this is the reason !!! We can ask user to install unix subsystem to see if that fixes the issue. Though I still have no ideas why Board Manager attempt to run such a command. It should only download & extract the zip file to correct location !!!!!

hathach, 2017-04-07:
I will test this out with a new windows 10 in virtual machine. But still, why it happens to this core only ????

microbuilder, 2017-04-07:
I've never seen this either ... I'll have a look through the relevant files here again to see if I spot something again.

microbuilder, 2017-04-07:
Perhaps because of 'mingw' here: https://github.com/adafruit/arduino-board-index/blob/gh-pages/package_adafruit_index.json#L1441

hathach, 2017-04-07:
possibly, not entirely sure though. But look like the potential cause

hathach, 2017-04-07:
Is there anyway we can test without changing that file online since it will affect all the users ???

hathach, 2017-04-08:
Hmm, I think you are right `i686-mingw32` look likes a 32bit or something. If Mingw is not installed, it falls back to other ARCH possibly Linux.

hathach, 2017-04-08:
I take that back. Look like `i686-mingw32` is normal, Arduino package json file also uses that. Must be something else :(

hathach, 2017-04-18:
caused by symlink in nrfjprog tool for linux & osx. Windows does not like those link much.

closed 2017-04-19 by hathach