adafruit/Adafruit_nRF52_Arduino #31: NFC Hardware support

hathach, 2017-04-09:


hathach, 2017-04-12:
https://devzone.nordicsemi.com/blogs/920/how-to-test-nrf52-nfc-tag-with-reader-adafruit-tag/

microbuilder, 2017-04-12:
This has already been tested during development and confirmed working, though adding demo code that works with the SD and Bluefruit API remains as a ToDo ... though I think improved Central support is more critical at this stage.

hathach, 2017-04-12:
Yeah, the issue reminds us to create some helper to make NFC using easier. But it is not urgent yet. Cental API is quite OK ( for single connection now).

hathach, 2018-11-15:
We still haven't worked on NFC just yet. SDK has a strict license, so we didn't use any of its files, only subset of nrfx module for some prph driver. Nordic SDK has quite a bit of dependencies, I am not sure but it may require some effort to include its files for development. If you could oob pairing and willing to submit an PR, we will be more than happy to merge. :)

hathach, 2020-02-03:
closed due to lack of activities

closed 2020-02-03 by hathach