adafruit/Adafruit_nRF52_Arduino #38: A seperate I/O service ?

hathach, 2017-04-11:
Control basic GPIO, execute command such as
- pinMode
- digitalWrite
- analogWrite
- analogRead
Etc....

neomanic, 2017-07-03:
I've got a basic example running that emulates the Nordic [LED Button Service](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk51.v10.0.0%2Fgroup__ble__sdk__srv__lbs.html
), since I ported an application over from an nRF51 dev board to the Bluefruit nRF52. If there's any interest I can share the code.

hathach, 2017-07-05:
@neomanic  that is great, unfortunately, but those code are likely limited by Nordic license right ?. Anyway, we haven't decided which characteristics and/or its data format aka specs yet.  This issue is just a reminder for us. We hope to have a simple service with minimal Chars to fully control basic IO (digital + analog). The purpose ease the full I/O control from mobile app without a custom parser based on bleuart (like neopixel).

monteslu, 2017-07-05:
I have some experience using both models from javascript/node using bleuart https://github.com/monteslu/ble-serial  + BLE firmata:  https://github.com/firmata/arduino/tree/master/examples/StandardFirmataBLE

as well as new GPIO service I've started here:  https://github.com/monteslu/ble-io with a firmaware for curie (Arduino 101 & TinyTile) so far.    There is also firmware by @jacobrosenthal for myNewt to implement this new gpio service pattern in the works.

The StandardFirmataBLE/bleuart route is still more feature complete at this point, but my ble-io firmware currently does handle digital and analog read and writes as well as servos.

I'm kinda partial to the newer io service pattern at this point.  Mostly because there's no state or extra handshaking that the client has to maintain.  You simply issue bytes to write to pins, or listen for notifications after putting pins in read mode.

I'll try and get some docs up soon, but it's basically a digital characteristic, a analog characteristic, and a config characteristic.  The config does thing like setting pin mode, or configuring servos.  Most of the bytes I'm using to specify those configurations are taken from the firamata protocol:  https://github.com/firmata/protocol/blob/master/protocol.md and my current arduino reference firmware is dependent on firmata libraries.

monteslu, 2017-07-05:
We used the ble-io firmware for this event:  https://www.citrix.com/blogs/2017/06/05/rockin-robots-synergy/

Worked out great.  The students connected to the bots using web bluetooth on chromebooks.

neomanic, 2017-07-06:
@hathach It's not using any Nordic code at all, just creating a similar Service+Characteristics. The only thing copied is the UUID. 

But I do really like @monteslu's approach. Would be nice to have a standardardised Arduino-IO control service+characteristics library that each BLE-providing board could then implement.

hathach, 2017-07-07:
hi @monteslu and @neomanic  That is great, I take a look at the firmata protocol, although it is well designed but based on MIDI message, and indeed add more works to both client as well as central. What I am thinking is a simple service with 2 simple chars Command and Data

- Command (write, write without resp): to execute client code such as digitalMode(pin, mode), where Mode can be directly enum from Arduino e.g INPUT = 0, OUTPUT = 1.
- Data (read, notify): retrun data from such as digitalRead(), analogRead(). 

A command ID can even mapped to an Arduino API, since that is standard, prph only need to pull the according function from table and then feed following data without much effort on parsing.

microbuilder, 2017-07-07:
We have the Pin I/O code for the nRF51 (https://github.com/adafruit/Adafruit_BLE_PinIO) and support for this in our Bluefruit LE Connect apps (under the Pin I/O tab), but haven't migrated this to the new nRF52 yet. Unfortunately, Firmata can be a PITA to get setup on the code side.

What Thach is thinking about is something simpler, and so easier to implement in apps by customers or when you don't need or want the full overhead of Firmata.

monteslu, 2017-07-07:
My new firmware isn't firmata, it's just happens to be using some firmata abstraction helpers under the covers so that we don't have to re-invent the wheel on things like supported pin modes across different boards. The one for myNewt that Jacob is working on doesn't use firmata libraries at all.

I've added some quick documentation for the javascript client side:  https://github.com/monteslu/ble-io

As well as some documentation on the the BLE IO service so far:  https://github.com/monteslu/ble-io/blob/master/service.md

I don't know a ton about the adafruit nrf52, but the service should be doable.

@microbuilder take a look, should be simpler than a firmata stream.

hathach, 2017-07-07:
hi @monteslu and @neomanic  That is great, I take a look at the firmata protocol as well, although it is well designed but based on MIDI message, and indeed add more works to both client as well as central. What I am thinking is a simple service with 2 simple chars Command and Data

- Command (write, write without resp): to execute client code such as digitalMode(pin, mode), where Mode can be directly enum from Arduino e.g INPUT = 0, OUTPUT = 1.
- Data (read, notify): retrun data from such as digitalRead(), analogRead(). 

A command ID can even mapped to an Arduino API, since that is standard, prph only need to pull the according function from table and then feed the rest of data as arguments without much effort on parsing.

hathach, 2017-07-07:
However, we (Adafruit) also need to implement firmata as separated libraries as well since other user may find that useful (and our current Bluefruit app is using it). #58 . While waiting to finalize spec for this one.

ladyada, 2017-07-07:
hiya poking in here! i know firmata is not perfect and it would be awesome to design a new protocol, but i'd like to recommend we stick to and only expand firmata. its very standardized in the maker world - and if we stick to it we have a better chance of working with software like Scratch, App Inventor, Code.org, etc. I'd rather add a firmata-parsing block to adafruit.io than have new protocol to support :)

monteslu, 2017-07-07:
@ladyada generally speaking I agree.  Firmata is perfect for generic IO and more complex things like i2c.  I've had success in streaming firmata over a whole ton of transports such as [tcp](https://github.com/mwittig/etherport-client), [udp](https://github.com/monteslu/udp-serial), [mqtt](https://github.com/monteslu/mqtt-serial), [socket.io](https://github.com/monteslu/socket.io-serial), [BLE](https://github.com/monteslu/ble-serial), [MIDI](https://github.com/monteslu/splidi-serial), and even [webusb](https://www.youtube.com/watch?v=Z1Nk2hH2wFE) .

Main issue for me is that the the Adafruit implementations have diverged enough from adurino/firmata upstream that it has been difficult for me to support [StandardFirmataBLE](https://github.com/firmata/arduino/tree/master/examples/StandardFirmataBLE) and use the nordic uart stream service to communicate with it.  I gave it a shot on the BlueFruit, and unfortunately don't have the C chops to resolve this.

We've spent a ton of time trying to make firmata better and I'd love to work this out :)

A separate issue is buffering streams on small boards.  With a generic IO service we don't need to do much buffering and parsing on the peripheral side.  Each command is a small transaction.

The generic IO BLE service I'm proposing makes use of services/characteristics directly and uses the same bytes specified in the firmata protocol.   This has been much more reliable for me especially over web bluetooth.

ladyada, 2017-07-07:
our underlying protocol has not diverged, just how we buffer data for sending over BLE (also we made it a little easier to set up boards, imo)
im totally ok with changing the buffering setup but i'm not convinced we need a new protocol :)

hathach, 2017-07-12:
Just want to say, we would also implement the Nordic Led Button service as optional libraries as well. Since Nordic make an app for Android/iOS . This is still useful for users that want to toggle control nrf52 as easily as possible #59

hathach, 2017-08-01:
close in favor of firmata #58  which is already done. Device can configured to have multiple bleuart service instant one for firmata, another for other purpose if needed. Thank you all for your great suggestions.

closed 2017-08-01 by hathach