adafruit/Adafruit_nRF52_Arduino #42: Move to softdevice s132 v3.0.0

dralves, 2017-04-16:
Hi!
  
  Thank you for all the hard work, bluefruit is awesome.
  I was wondering: Is there a plan to move to version 3.0.0 of the s132 sdk?
  That version is supposed to suppose BLE packet length extension and thus increase the possible throughput by a lot.

Best

microbuilder, 2017-04-16:
Migrating to a new SD is problematic since the SD is part of the bootloader and is flashed during the manufacturing process via an SWD debugger, which most customers won't have access to. OTA updates are a possibility for the SD, though not risk free and this creates a situation where we can't guarantee that every user is on the same SD, and the SDK code and Bluefruit API is generally tightly coupled to SD releases. :/

We might do this in a future release and tick the board versions up one to have a clear separation between boards with 3.0.0 and the currently shipping release, but at the moment we are focusing our energy on completing the API and adding new features.

It's on our radar, though!

closed 2017-04-16 by microbuilder

hathach, 2017-04-16:
The reason we went to sdk 11 + SD v2 is the lack of serial bootloader support in the current secure DFU. Both button less OTA + Serial DFU are in experimental phase of current sdk 13. Btw, the SD v4 is already out, v5 is in alpha development as well. There is no point to upgrade to v3 anymore. Let's wait for SD v5 that come with the solid serial DFU support.

Don't worry about the future, we have testing upgrading Bootloader + SDv3 internally from current combo. It is doable, just one-way path. The bootloader is purposely given large enough size ~40KB to handle both serial + OTA + secure DFU in the future. However, it is still dangerous process, and has little benefit now ( without serial DFU).

hathach, 2017-04-18:
for reference #46

dralves, 2017-04-18:
Thank you for the explanation!
I'll follow progress closely.
If there is a need for beta, or even alpha, testers, let me know :)

bossesand, 2017-05-17:
As I have a  SWD debugger and want to update to s132 v4 I wonder what the needed action is?
Is it enough to flash S132 v4.0.3 int to the feather and substitute the new version of the .h files ?

microbuilder, 2017-05-17:
You will also need to deal with the breaking changes in the peripheral library and helper classes and Nordic SDK. It isn't just a matter of 'I'll just flash this, and problem solved', alas. You're of course free to poke at it, but it's probably less trivial than you are expecting.