adafruit/Adafruit_nRF52_Arduino #50: Custom 128 bit UUIDs bytes reversed

ipv6wiz, 2017-05-11:
Hello,

This variable definition 
uint8_t custom_service_uuid[]           =  {0xa4, 0x95, 0xff, 0x20, 0xc5,0xb1,0x4b,0x44,0xb5,0x12,0x13,0x70,0xf0,0x2d,0x74,0xde};

Yields this transmitted UUID when compiled with the Adafruit Feather 32u4 library. This one matches the orientation of the UUID bytes on other BLE devices that I have programmed, such as the Light Blue Bean and the SensorTag.
Service UUID :  A495FF20-C5B1-4B44-B512-1370F02D74DE

And this reversed transmitted UUID when compiled with the new Adafruit Bluefruit nrf52 library. This one is definitely reversing the bytes which yields an incorrect UUID.
Service UUID :  DE742DF0-7013-12B5-444B-B1C520FF95A4

Please advise on when a fix will be applied, I am sure that this impact others as well.

Best regards
John

hathach, 2017-05-12:
The little endian inversion is native API from SoftDevive API. For nrf51 series, we did an extra effort to convert from big endian to little. For nrf52, since the code run on the chip without another MCU, advanced user can call SoftDevice API directly, so we leave it as it is (little endian). 

Admittedly, this could cause some inconveniences, we will add an function to change the endian big <-> little for uuid128 in the next release

hathach, 2017-06-22:
taking a look at the code again. It is better to use little endian for uuid128 since that is the native format supported by SoftDevice. We will make sure it is documented on the learning guide

closed 2017-06-22 by hathach