adafruit/Adafruit_nRF52_Arduino #51: Power consumption

lpercifield, 2017-05-15:
First of all, awesome work on this hardware and core!

Secondly, Power consumption:
I've been testing with this core, and it seems that there is significant quiescent power consumption, regardless of actual hardware. 
On the Feather nRF52 board using the beacon example I'm seeing 13mA average, and thats with the connection LED disabled. 
Using a Red Bear Nano 2 (with their bootloader) but the same compiled hex, and powering off the VDD pin (3.3v) I'm still seeing ~6mA.

Using the Red Bear Nano 2 and their beacon code, I can get average power consumption below 20uA!

I'm hoping to dive in and do some deep level debugging with this core, but need to get the power under control relatively quickly for a prototype.

Any thoughts or suggestions would be helpful!

Thanks and keep up the great work

microbuilder, 2017-05-15:
Getting proper low power numbers is high on our ToDo list, but we've made a conscious choice to focus on critical BLE features and helper classes first since that's what 95% of people 'need' out of the gate. 

Low power is unfortunately a very time consuming thing to get right since everything has to be done in just the right fashion to get the best case scenario power numbers, and it's 90% a firmware problem.

Feel free to open a PR or issue if you have something specific to contribute or point out.

lpercifield, 2017-05-15:
Great, I'll be putting in some hours on it this week, so I'll see what I can find.

microbuilder, 2017-05-15:
You mind find this a useful purchase if you're serious about low power measurements but don't have access to a higher end DMM or dedicated low power equipment: http://www.nordicsemi.com/Products/Power-Profiler-Kit

There is an alternate cross-platform UI for it as well if you aren't on Windows: https://qoitech.com/#/nordic?_k=1xkxah

bluebrolly, 2018-01-18:
I am interested in this as well. Hopefully this will be sorted out soon

chaffneue, 2018-02-07:
Think it's something to do with Arduino timer interrupts not letting the MCU enter low power mode? Any idea where to start looking into this? I'm seeing about the same, a consistent 13mA draw no matter if I'm using waitforevent() or not. Definitely seems like the power consumption hikes up in the loop section. Here's a video from my startup sequence - It goes from about 7mA to 13mA. https://www.flickr.com/gp/chaffneue/mUp94h

chaffneue, 2018-02-08:
Oh. It's a hardware issue with this board https://www.adafruit.com/product/3406 (It's out of stock now).. damn what a waste. For anyone stuck on this, you can remove the USB bridge chip on this model pretty easily with hot air, but programming would need an external serial board connected to the rxd/txd pins  :/ I guess I'll just wait for the new hardware and just use this as a dev board in the meantime. I was kind of wondering if that would happen looking at the schematics. I was assuming the 2104 would be bus powered only and out of the circuit when powered by the lipoly, but no. 

https://forums.adafruit.com/viewtopic.php?f=24&t=128823&p=641636#p641352

https://devzone.nordicsemi.com/f/nordic-q-a/28911/can-t-put-adafruit-nrf52-feather-board-into-low-power

ladyada, 2018-02-08:
yep that was our mistaken assumption too. the next time we have the nrf52 feather in stock will be the new version, please sign up and you'll be notified (no ETA, so please sign up! :)

chaffneue, 2018-02-08:
Signed up! :)

ladyada, 2018-02-08:
OK we might have a *few* stragglers go into stock, if you get a notification in the next day - its just the last of the stock (we do a final count and often there's a couple extra kicking around)

chaffneue, 2018-02-10:
I've removed the CP2104 from the board and there's still about a 1mA constant current draw from somewhere. I'm guessing it's the quiescent current from the regulator. Will the new design have tweaks to the power supply as well?

ladyada, 2018-02-10:
nope, just the cp2104 is being changed in this revision. current regulator is ap2112

pbecchi, 2018-02-27:
I have found that the 7mA are pin related.
I guess there is some pin setting, may be coming from bootloader or from the libraries(Nffs?) ,that may not be compatible with my board hw! I have not found yet the actual pins that may draw this current or the way to take it out, but it doesn't come from my code.

coltonottley, 2018-07-19:
There's definitely some problem with the arduino libraries, it's not the hardware. Using the Nordic SDK I can get the power consumption down to 16 uA while advertising. Doing the same thing with the Arduino IDE I'm seeing a constant 7mA power consumption once BLE has been connected or once the code is allowed to enter loop.

jpconstantineau, 2018-07-19:
Adding "Delays" in the loop does lower power consumption significantly. I had 6-12mA before I changed a few things in my code but the biggest drop happened when I added "delay(25);" to my loop to reduce the number of times it cycles though the loop per second.  I now have around 500uA consumption.  Nowhere close to the 10-20uA while advertising but an improvement.

coltonottley, 2018-07-19:
Yeah that's a good find. That works for me as well. Thanks for sharing. It looks like delay is using the freeRTOS vTaskDelay function then? And loop is a scheduled task.

pbecchi, 2018-07-19:
I agree!  
The only way I have found to lower power consumption is to insert a delay(Ms) in the loop().
Increasing Ms lower the power drain.....
From 6mA with Ms=0
To     100uA with Ms=200.

Ironman-007, 2022-07-20:
@pbecchi : what are Nffs statements? Thanks!