adafruit/Adafruit_nRF52_Arduino #52: Bonded keys are not being saved

afriggeri, 2017-06-04:
Follow up on [this thread](https://forums.adafruit.com/viewtopic.php?f=57&t=117099), I did some digging and here is what I found: the [following](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/9b5154564458ec8e031274ddd1e5fd82f195b382/libraries/Bluefruit52Lib/src/bluefruit.cpp#L698) call fails, which means that the keys aren't properly saved. Going deeper in the stack, it looks like the call to [`fs_write` here](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/9b5154564458ec8e031274ddd1e5fd82f195b382/libraries/nffs/src/NffsFile.cpp#L161) returns `FS_EACCESS` which seems weird because the file was opened with `FS_ACCESS_WRITE`. 

@hathach: this doesn't look like an OSX problem, I don't have an android device to replicate this unfortunately. Any idea how to fix this?

afriggeri, 2017-06-04:
More digging, looks like the filesystem is somehow screwed on my feather:

1. When running the clearbonds example, it actually doesn't clear the bonds, the call to `fs_unlink` fails with `FS_EOFFSET `

2. The call to `fs_open` [here](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/9b5154564458ec8e031274ddd1e5fd82f195b382/libraries/nffs/src/NffsFile.cpp#L63) when opening the file to store the bond keys fails with `FS_EOFFSET ` too.

Is there anyway I can entirely reset the board?

afriggeri, 2017-06-04:
Ok, do I grounded FRST to erase the memory, and now I don't have this error anymore. I'm still not sure why my FS got into that weird state

hathach, 2017-06-05:
I am about to say FRST as well. I am glad that you could get it work finally. Apparently NFFS file node is corrupted somehow. We have no ideas as well, NFFS is new to us as well. However, the code should report a meaningful error for this, so that we are not totally in blind. We will update error message in the next release.

PS: there is Nffs.format() as well to clear the whole file system while leaving sketch code. https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/9b5154564458ec8e031274ddd1e5fd82f195b382/libraries/nffs/src/Nffs.h#L61

hathach, 2017-06-22:
related to https://forums.adafruit.com/viewtopic.php?f=57&t=119210

hathach, 2017-06-22:
This example use low level sd_flash_erase() API with SoftDevice disabled. Which is the same way as bootloader perform factory reset. It should work if your fs got corrupted again but somehow still pass sanity check (possibly a bad/failed inode)

closed 2017-06-22 by hathach