adafruit/Adafruit_nRF52_Arduino #57: Apple homekit protocol for non-commercial version

hathach, 2017-06-27:
Apple just recently open homekit protocol for non-commercial version. 
https://developer.apple.com/support/homekit-accessory-protocol/

hathach, 2017-10-05:
require #46  since homekit requires ATT MTU at least 100 bytes, recommended 150 or larger.

hathach, 2017-10-27:
srp based on mbedtls https://github.com/dwimberger/mbedtls-csrp
mbedtls https://github.com/ARMmbed/mbedtls
mbedtls also support curve 25519

renssies, 2018-11-05:
Any update on the implementation of HomeKit. Really looking forward to making battery powered BLE HomeKit sensors! 👍

hathach, 2018-11-06:
lack of manpower, we only manage to complete the security process which is the most complicated procedure in the whole homekit specs. We will resume the development later on.

renssies, 2018-11-06:
Too bad there is not enough manpower to finish it soon. Unfortunately, I don't know enough about the HAP BLE spec in order to help.

mharper, 2018-12-30:
@hathach if you want to give me a vector on how I can help here, I'll do my best!

mharper, 2018-12-31:
_FINALLY_ got my Apple TV 4 setup as a Home Hub. Will start looking into how to move forward here.

hathach, 2019-01-07:
@mharper thank you for your willingness. There is still a lots of work to get homekit protocol. We only passsed the pairing security procedure, the whole HAP protocol needed to be implemented with session key (verify procedure). 

PS: Currently the homekit build failed due to the migration from Nffs to littlefs. We will update to fix build later on.

renssies, 2019-04-29:
@hathach I'm guessing this issue was accidentally closed?

hathach, 2019-04-29:
@renssies indeed, it is an accident :). Thank you for poiting it out

renssies, 2019-12-19:
This might be of great help implementing HomeKit. Just released  today: https://github.com/apple/HomeKitADK

renssies, 2019-12-19:
Also, Apple has released a new Non-commercial HAP spec, which release contains a new Bluetooth LE implementation with a faster response time: https://developer.apple.com/homekit/specification/

hathach, 2019-12-19:
thanks you for updating us with latest homekit changes. Unfortunately, I currently have no bandwidth to work on this. There is still months of works before we could get something useful running with Arduino nrf5x.

renssies, 2019-12-19:
No problem, it was more to let others know that might want to help, because the issue is marked as “Help wanted” :)

AGRO666, 2023-07-07:
Is there any news on the implementation of the HomeKit?

hathach, 2023-07-07:
there is no time to work on this, and no new progress

commit 3a91089a (2019-04-27): fix #57 left over from hid keycode <-> anscii