adafruit/Adafruit_nRF52_Arduino #58: firmata libraries

hathach, 2017-07-07:
https://github.com/adafruit/Adafruit_BLE_PinIO

microbuilder, 2017-07-07:
ToDo: The Pin I/O functionality in the Bluefruit LE Connect apps relies on Firmata, which needs to be ported to the nRF52832 and this codebase (since the API is completely different).

hathach, 2017-07-24:
https://github.com/firmata/protocol
https://github.com/firmata/arduino

hathach, 2017-07-24:
need #58 to run StandardFirmata example

hathach, 2017-07-25:
kind of work, but I got issue with capability query. Googling

[image]

hathach, 2017-07-25:
issue seems to be bleuart.write() is an notification, while firmata use a lot of one byte write --> local buffered needed, somewhat similar to what Todd did with blemidi.

ladyada, 2017-07-26:
dont forget to look at https://github.com/adafruit/Adafruit_BLE_PinIO which is the port i did for the nrf51 boards :)

hathach, 2017-07-26:
Yeah, I look at that repo a lot for reference :)

hathach, 2017-07-26:
added function to enable bufferTXD() to deal with several write(1) from firmata. Allowing us to use bleuart with official code . Only need to submit a pull request for boards.h https://github.com/firmata/arduino/blob/master/Boards.h

Though it is relatively slow. tweaking it.

hathach, 2017-07-27:
update: huge amount of data is caused by command to configure I/O as INPUT --> floating pin --> data change randomly. Fixed by forcing INPUT_PULLUP in set calblack

hathach, 2017-07-27:
The port is complete, the example is almost identical to stock version, except for Bluefruit configure

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Bluefruit52Lib/examples/Peripheral/StandardFirmata/StandardFirmata.ino

and use bleuart as transportation

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Bluefruit52Lib/examples/Peripheral/StandardFirmata/StandardFirmata.ino#L789

The Firmata library is added but untouched. The only modification is Boards.h to add our board support
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Firmata/Boards.h#L785

when running stable enough, we will submit a pull request

hathach, 2017-07-27:
Although Boards.h declare all digital has PWM
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Firmata/Boards.h#L785

But Bluefruit App only show PWM capability for Pin 2-5 and 28-31. May need some check later, it is easier to test PWM with pin17, pin 19 (connected to LED1 & LED2)

[image]

[image]

hathach, 2017-08-01:
Finally, I catch the issue, it is part of the App parser for Query Capabilities, maybe due to the lower spec of firmata v 2.3 when the application is implemented. The current version is 2.5

in query capabilities response

https://github.com/firmata/protocol/blob/master/protocol.md#capability-query

the standard firmata in addition to report INPUT, also report INPUT_PULLUP (0x0b) as well

However, the parser does not include INPUT_PULLUP in the switch

https://github.com/adafruit/Bluefruit_LE_Connect_Android/blob/master/app/src/main/java/com/adafruit/bluefruit/le/connect/app/PinIOActivity.java#L408

E.g a pin response as follow is valid

00 01 0B 01 01 01 02 0A 03 08 04 0E 7F

- 00 01 -> input
- 0B 01 -> input pull up
- 01 01 -> out put
- 02 0A -> analog read with 10 bit resolution
- 03 08 -> pwm with 8 bit resolution
- 04 0E -> Servo
- 7F end byte

when parser reach 0B, it skips and only increase i by 1 --> cause mis-match the rest of the data.
In my opinion, the specs seems to imply that every pin must follow by 2 bytes ( mode + resolution ) regardless of the mode. The parser should always increase by 2 each loop.

PS: iOS should be the same as well.

hathach, 2017-08-01:
2nd issue with Bluefruit app, we need to use `Extended Analog` to support PWM/AnalogRead  on pin > 15 which is the case of nrf52 (e.g LED1 is 17, LED2 is 19)
https://github.com/firmata/protocol/blob/master/protocol.md#extended-analog

https://github.com/adafruit/Bluefruit_LE_Connect_Android/blob/master/app/src/main/java/com/adafruit/bluefruit/le/connect/app/PinIOActivity.java#L592

closed 2017-08-15 by hathach