adafruit/Adafruit_nRF52_Arduino #61: Advanced Central mode advertising packet demo

microbuilder, 2017-07-15:
A sample sketch should be added for Central mode similar to [adv_advanced.ino](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Bluefruit52Lib/examples/Peripheral/adv_advanced/adv_advanced.ino) showing the following features:

## Requirements

- [ ] Clearly document the code on how to setup the scanner, as well as how to handle both normal adv packets and scan response (active scanner) packets
- [ ] Show how the scanner frequency can be adjusted

When a packet arrives (adv or scan response):

- [ ] Display the timestamp and basic device details (RSSI, address, addr type, adv type) every time a packet is received
- [ ] Print the raw contents of the advertising packet in zero-padded HEX format
- [ ] Parse the packet contents for the following information:
  - [ ] Device name (display if prsent)
  - [ ] 16-bit UUID list
  - [ ] 128-bit UUID list
  - [ ] BLE UART service found (True/False)
  - [ ] DIS found (True/False)
  - [ ] Check for manufacturer specific data and display the payload if present
- [ ] Enable active scan for scan response packets
  - [ ] When a scan response packet arrives, display it the same way as the main adv packet, but making it clear this is the scan response packet
- [ ] New line between packets to help parsing and add some visual white space

## Output

This can be updated as needed, but a clean text output of the packet contents should be produced similar to the following text:

```
[00:11:22.3333] ADV packet received from AA:BB:CC:DD:EE:FF
        PAYLOAD 24 bytes
                00 11 22 33 44 55 66 77 88 99 AA BB CC DD EE FF
                00 11 22 33 44 55 66 77
           RSSI -37 dBm
      ADDR TYPE Random
       ADV TYPE Connectable
    DEVICE NAME Display here if present (otherwise skip this line)
   16-BIT UUIDS AABB, CCDD
  128-BIT UUIDS 58D5E212-165B-4CA0-909B-C86B9CEE0111
       BLE UART UUID Found
            DIS UUID Found
MANUF SPEC DATA 7 bytes, Company ID: 0x1234
                00 11 22 33 44 55 66

[00:11:22.3675] SCAN RESPONSE packet received from AA:BB:CC:DD:EE:FF
                ...
```

hathach, 2017-07-15:
great idea, we could update the current central_scan demo.

microbuilder, 2017-07-15:
Thankfully we have printf available with proper string formatting, which will be a life saver here. :)

I just think this will be the best documentation in that it will be kept up to date with any API changes, and like the peripheral mode `adv_*` sketches people can see how things work in the real world more easily here.

It's a useful debug tool as well.

hathach, 2017-07-15:
we may need to set a slow scanning interval, currently it is too fast with several package received per second. Since we don't connect, it keeps coming.

microbuilder, 2017-07-15:
I'll write some helper functions to parse and spit out the verbose output, and we can default to just the raw payload and timestamp which it should be able to handle with light traffic. 

We can do the full verbose output once when a new MAC address is encountered for ADV and the SCAN RESPONSE, since it's unlikely to change.

microbuilder, 2017-07-17:
Just playing with this. We need some helpers to parse the data, but the following code is an OKish bare-bones skeleton:

```
/*********************************************************************
 This is an example for our nRF52 based Bluefruit LE modules

 Pick one up today in the adafruit shop!

 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

#include <bluefruit.h>

void setup() 
{
  Serial.begin(115200);

  Serial.println("Bluefruit52 Central ADV Scan Example");

  /* Enable both peripheral and central modes */
  Bluefruit.begin(true, true);

  /* Set the device name */
  Bluefruit.setName("Bluefruit52");

  /* Set the LED interval for blinky pattern on BLUE LED */
  Bluefruit.setConnLedInterval(250);

  /* Start Central Scanning
   * - Enable auto scan if disconnected
   * - Interval = 100 ms, window = 80 ms
   * - Use active scan (used to retrieve the optional scan response adv packet)
   * - Start(0) = will scan forever since no timeout is given
   */
  Bluefruit.Scanner.setRxCallback(scan_callback);
  Bluefruit.Scanner.restartOnDisconnect(true);
  Bluefruit.Scanner.setInterval(160, 80);       // in units of 0.625 ms
  Bluefruit.Scanner.useActiveScan(true);        // Request scan response data
  Bluefruit.Scanner.start(0);                   // 0 = Don't stop scanning after n seconds

  Serial.println("Scanning ...");
}

void scan_callback(ble_gap_evt_adv_report_t* report)
{
  /* Display the timestamp and device address */
  if (report->scan_rsp)
  {
    Serial.printf("[%12d] SCAN RESPONSE packet received from ", millis());
  }
  else
  {
    Serial.printf("[%12d] ADV packet received from ", millis());
  }
  Serial.printBuffer(report->peer_addr.addr, 6, ':');
  Serial.print("\n");

  /* Raw buffer contents */
  Serial.printf("%14s %d bytes\n", "PAYLOAD", report->dlen);
  Serial.printf("%15s", " ");
  Serial.printBuffer(report->data, report->dlen, '-');
  Serial.println();

  /* RSSI value */
  Serial.printf("%14s %d dBm\n", "RSSI", report->rssi);

  /* Adv Type */
  Serial.printf("%14s ", "ADV TYPE");
  switch (report->type)
  {
    case BLE_GAP_ADV_TYPE_ADV_IND:
      Serial.printf("Connectable undirected\n");
      break;
    case BLE_GAP_ADV_TYPE_ADV_DIRECT_IND:
      Serial.printf("Connectable directed\n");
      break;
    case BLE_GAP_ADV_TYPE_ADV_SCAN_IND:
      Serial.printf("Scannable undirected\n");
      break;
    case BLE_GAP_ADV_TYPE_ADV_NONCONN_IND:
      Serial.printf("Non-connectable undirected\n");
      break;
  }

  /* Device Name */
  Serial.printf("%14s %s\n", "DEVICE NAME", "ToDo!");

  /* Check for BLE UART UUID */
  Serial.printf("%14s ", "BLE UART");
  if ( Bluefruit.Scanner.checkReportForUuid(report, BLEUART_UUID_SERVICE) )
  {
    Serial.printf("UUID Found\n");
  }
  else
  {
    Serial.printf("Not Found!\n");
  }

  /* Check for DIS UUID */
  Serial.printf("%14s ", "DIS");
  if ( Bluefruit.Scanner.checkReportForUuid(report, UUID16_SVC_DEVICE_INFORMATION) )
  {
    Serial.printf("UUID Found\n");
  }
  else
  {
    Serial.printf("Not Found!\n");
  }

  Serial.println();
}

void loop() 
{
  // Toggle both LEDs every 1 second
  digitalToggle(LED_RED);

  delay(1000);
}
```

This will give you something like this in the output:

```
[      350845] ADV packet received from DF:5F:24:31:4A:D2
       PAYLOAD 24 bytes
               02-01-06-02-0A-00-11-06-9E-CA-DC-24-0E-E5-A9-E0-93-F3-A3-B5-01-00-40-6E
          RSSI -70 dBm
      ADV TYPE Connectable undirected
   DEVICE NAME ToDo!
      BLE UART UUID Found
           DIS Not Found!

[      350873] SCAN RESPONSE packet received from DF:5F:24:31:4A:D2
       PAYLOAD 23 bytes
               16-09-41-64-61-66-72-75-69-74-20-42-6C-75-65-66-72-75-69-74-20-4C-45
          RSSI -69 dBm
      ADV TYPE Connectable undirected
   DEVICE NAME ToDo!
      BLE UART Not Found!
           DIS Not Found!
```

hathach, 2017-07-17:
There is a helper parser that you can use to get the payload based on the type, which you can use to retrieve the device name if advertised 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Bluefruit52Lib/src/BLEScanner.h#L70

microbuilder, 2017-07-17:
Thanks for the heads up ... I've pushed a demo out here that shows most of the API functions and allows people to parser advertising packets in a reasonably complete manner: https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/Bluefruit52Lib/examples/Central/central_scan_advanced/central_scan_advanced.ino

closed 2017-07-28 by microbuilder