adafruit/Adafruit_nRF52_Arduino #63: Add RSSI filter to scan results

microbuilder, 2017-07-17:
When we start the scanning process, we should add an optional argument to specify an RSSI filter. 

Any value <= the specified RSSI value will be ignored and the callback won't be fired. This will make it easier to only see data from devices that are nearby.

This can either be an optional argument in `Bluefruit.Scanner.start(...)` or or a new function like `Bluefruit.Scanner.setRSSIFilter(int8_t min_rssi);`.

hathach, 2017-07-17:
great idea, though, user can also simply write an 
if rssi < value

in the callback and return as well.

hathach, 2017-07-17:
hmm, although it is doable in user callback, maybe we could extend the filter a bit e.g name contains a specific string and/or having uuid16 or uuid128. It is all about convenience.

microbuilder, 2017-07-17:
I'd also like to filter by other values like the HW address, though, and having some filtering behind the scenes might keep code cleaner. This isn't urgent, though, I'm just filing it as a 'ToDo' and as the API gets the tires kicked in we can come back to it.

closed 2017-07-18 by hathach