adafruit/Adafruit_nRF52_Arduino #64: Bluetooth LE Mesh

hathach, 2017-07-19:
https://www.bluetooth.com/what-is-bluetooth-technology/how-it-works/le-mesh

may requires BLE5 SoftDevice #46

microbuilder, 2017-07-19:
Thanks for filing an issue.

I was reading the 1.0 spec last night, and it's limited since it's based entirely on advertising (which is due to a design limitation of BLE of course and the nature of connections), but I still think this is useful and it's good to have a standard for mesh based on inexpensive BLE devices since they are so plentiful, and unlike 802.15.4 they are generally compatible across manufacturers (though 15.4 is a much better basis for mesh, since it has reliable communication and you know when a message fails, etc.).

We should put a simple demo together for this down the road, though, since it could be useful in a home environment, and the nRF52 is reasonably affordable and allows you to connect to your mobile phone as a gateway or edge router to the sensor nodes.

hathach, 2017-07-20:
sdk for mesh http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF5-SDK-for-Mesh

microbuilder, 2017-07-27:
https://devzone.nordicsemi.com/blogs/1147/introducing-nrf5-sdk-for-mesh/

microbuilder, 2017-11-09:
Yeah, there's a misconception (that I had myself at first since the BLE Mesh announcement was timed with BLE5) that this is tied to the latest BLE spec, but that definitely isn't the case. Thanks for tossing that out to clarify though!

hathach, 2021-06-03:
closed since it is not relevant anymore

closed 2021-06-03 by hathach

Krastanov, 2021-06-03:
@hathach , for reference could you say why it is not relevant anymore? I am subscribed for some updates on this project because I use it once a year and having mesh would make my life easier, but I have not tracked its progress in detail.

Is the project deprecated? Is mesh implemented somewhere else? Is this a "feature that will never be implemented"? A quick search through code and documentation did not show anything.

hathach, 2021-06-03:
it is not on the plan anymore.