adafruit/Adafruit_nRF52_Arduino #67: Issue with xTaskCreate and vTaskStartScheduler

netkruzer, 2017-07-21:
I'm trying to get my Adafruit Feather nRF52 to run separate tasks but the processor keeps restarting on it's own. I call the vAFunction() in my setup routine, but it never gets into the loop, and never calls the function vTaskCode.

`#define STACK_SIZE 512*3
int taskCheck = 0;
int taskCheckLoop = 0;
void setup() {

  Serial.begin(115200);
  Serial.println("Program Starting");
  vAFunction();
}

void loop(){
  Serial.print("TaskCheck "); // never gets here.
  Serial.print(taskCheck);

  Serial.print(" TaskCheck Loop");
  Serial.print(taskCheckLoop);
}

void vAFunction( void ){
     TaskHandle_t xHandle = NULL;
     // Tasks can be created before or after starting the RTOS scheduler
     xTaskCreate( vTaskCode,
                  "NAME",
                  STACK_SIZE,
                  (void *)1 ,
                  tskIDLE_PRIORITY,
                  &xHandle );
     Serial.print("Task Created\t\t");
     taskCheck++;
     Serial.println(taskCheck);
     Serial.print("eTaskGetState():\t");
     Serial.println(eTaskGetState(xHandle));
     // Start the real time scheduler.
     vTaskStartScheduler();
     Serial.println("Task Created Complete");
     // Will not get here unless there is insufficient RAM.
}
/* Task to be created. */
void vTaskCode( void * pvParameters )
{
    /* The parameter value is expected to be 1 as 1 is passed in the
    pvParameters value in the call to xTaskCreate() below. */
    configASSERT( ( ( uint32_t ) pvParameters ) == 1 );
    
    Serial.println("Task Executed");
    for( ;; )
    {
        taskCheckLoop++;
        
        //vTaskEndScheduler();
    }
}
`

Any idea what I'm doing wrong? Any idea how to debug these errors?

hathach, 2017-07-22:
Don't the scheduler, it is already started by main.cpp in the core folder.

netkruzer, 2017-07-22:
Ah, so I tried not using the scheduler in my Arduino program, instead modifying the one in the main.cpp to have 2 loop calls, loop and loop2. Then I just put my code I need to run parallel in loop2 and it mostly works. Not perfect, but better than I was getting.

Is there a way to do this correctly?

hathach, 2017-07-22:
No, I mean the scheduler is already started by main() function. You simply create the task, and skip the vTaskStartScheduler() call. Alternatively you can use our helper class Scheduler which implement the standard. startLoop()

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/rtos_scheduler/rtos_scheduler.ino

PS: Github issue is meant to report bug and request feature. If you still have issue please open a topic on support forum

closed 2017-07-22 by hathach