adafruit/Adafruit_nRF52_Arduino #69: support Tone()

hathach, 2017-07-24:
like #68

mennamorato, 2018-02-21:
@hathach When will Tone library be added to nRF52 for Arduino?  Thanks

hathach, 2018-02-22:
No ETA at all :)

mogar, 2019-04-14:
I second this request. If you have any pointers on getting started with implementing, I could work on a pull-request.

hathach, 2019-04-19:
@mogar  may be primo could help https://github.com/arduino-org/arduino-core-nrf52/blob/master/cores/arduino/Tone.cpp

ladyada, 2019-06-15:
@hathach we'll need this for CPB - lets implement soon, it's done with a timer not PWM

hathach, 2019-06-18:
@ladyada please try with master branch to see if that works

ladyada, 2019-06-18:
hmm tried this code on feather nrf52480 but no PWM on pin 5... does it work for you?

```
void setup() {

}

void loop() {
  pinMode(5, OUTPUT);
  tone(5, 4000, 100);
  delay(100);
}
```

hathach, 2019-06-19:
@ladyada ah sorry, I somehow delete the line that trigger pwm by mistake (a very important line of code). please pull from master and try again

ladyada, 2019-06-19:
done, tested - works great! thanks :)

closed 2019-06-19 by ladyada