adafruit/Adafruit_nRF52_Arduino #74: Nordic CLI Example

microbuilder, 2017-08-04:
Add an example of using the Nordic CLI over NUS, UART and RTT (configurable): https://devzone.nordicsemi.com/blogs/1150/new-command-line-interface-part-1/

This requires an update to SDK 14.0.0 or higher, though.

hathach, 2017-08-04:
that would need to quite some porting effort since the Nordic CLI firmware is tightly coupled with their sdk files.

microbuilder, 2017-08-04:
I just saw it on the blog, but haven't looked at the code in any detail (beyond the high mecro level). It's worth looking at down the road, though, which is why I filed an issue so it doesn't slip off the radar later.

hathach, 2017-08-04:
Yeah currently, their cli is still limited, but it is quite nice to have

jrzeszutko, 2018-02-08:
Hi,
May you please let me know what are main limitations for you in Nordic CLI? I am working on that and I could implement some improvements.

microbuilder, 2018-02-08:
If you want to submit a PR implementing this, we'll be happy to review it! It isn't something we're actively working on internally at the moment.

jrzeszutko, 2018-02-08:
I was more thinking about implementation on Nordic side. If you have any proposals I could pull it in to the next SDK release.

microbuilder, 2018-02-08:
Ah, sorry, I didn't notice you work at Nordic. :P I'll discuss this issue internally and see where we are with it, though we're currently busy with HomeKit and updates for the nRF52840.

hathach, 2020-02-03:
closed due to lack of activities

closed 2020-02-03 by hathach