adafruit/Adafruit_nRF52_Arduino #75: software serial data incorrect when advertising started

Zumpo, 2017-08-26:
Thanks for your awesome work on the Adafruit Feather nRF52. I really enjoy playing with it!

For my current project I try to combine the Softwareserial example with the Bleuart example to forward data I receive from my Ultimate GPS Breakout V3 to the Bluefruit app. Both examples you provide work fine when I use them separately, but not when I try to combine them. As soon as I start advertising, the data I receive from the GPS board gets a little scrambled. Things get much worse, when i connect to the feather from the bluefruit app. Any ideas?

Please find my sketch attached. Thanks so much for your help!
[nrf52_gps.zip](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/1253719/nrf52_gps.zip)

hathach, 2017-08-26:
Thanks for your kind words. Though github issue is for bug and feature tracker only. For other support please post on the support forum.

hathach, 2017-09-21:
in short, Bluefruit.begin() must be called before softserial.begin() for it to work

closed 2017-09-21 by hathach