adafruit/Adafruit_nRF52_Arduino #77: Add QEI Helper Class

microbuilder, 2017-09-11:
A helper class should be added for HW control of a simple rotary/quadrature encoder, tracking the absolute position, recent changes, as well as optionally the 'switch' on encoders that support it.

microbuilder, 2017-09-11:
We can do something like this:

```
// Position functions
void qei.clearAbs(void);
int32_t qei.getAbsPos(void);
int32_t qei.getDelta(int32_t *ccw, int32_t *ccw); // Returns 'total' changes since last call to .getDelta (cw + ccw), and optionally allows reading the cw (clockwise) and ccw (counter-clockwise) tickes individualls, which is more detail than the combined results returned by the function

// Callbacks
void qei.setQeiCallback(int32_t cw, int32_t ccw); // cw or ccw tick callback
void qei.setSwitchCallback(uint8_t state); // swtich callback if switch is present
```
The constructor should optionally accept a GPIO pin if the switch is present and init and handle that accordingly.

hathach, 2017-09-25:
both hw and sw encoder are added. It is a bit buggy but will be addressed by a separated issue.

commit 6786b468 (2017-09-15): close #77
closed 2017-09-25 by hathach