adafruit/Adafruit_nRF52_Arduino #78: OCF support

hathach, 2017-09-26:
related to #37 
reference

- https://openconnectivity.org/
- https://www.iotivity.org/documentation

microbuilder, 2017-09-26:
IoTivity is an open source implementation of the OFC standard, and is probably the way to go here. As a sidenote, I believe Mynewt support is still based on the 1.11 release of IoTivity, though, so we may want to remain consistent with that codebase.

microbuilder, 2017-09-26:
Here is a link to the Mynewt OIC code, just as a reference: https://github.com/apache/mynewt-core/tree/master/net/oic

NOTE: OIC has merged with OCF ... and is now called 'OCF', but you still run into the old 'OIC' name a lot since the merge was quite recent.

hathach, 2020-02-03:
closed due to lack of activities

closed 2020-02-03 by hathach