adafruit/Adafruit_nRF52_Arduino #82: Bluetooth 5 Long Range support

hathach, 2017-10-10:
**Longer range**

Bluetooth 5 introduced significant changes at the physical (PHY) layer for Bluetooth radio operation. Four data rates are now available 2Mbps, 1Mbps, 500kbps and 125kbps. The 2Mbps clearly offers higher throughput possibilities. Why offer lower datarates of 500kbps and 125kbps some may ask. The reason for these is that at these lower datarates and with the associated LE Coded processing using Forward Error Correction (FEC) it becomes easier to retirieve data transmissions in surrounding noise. In short, the sensitivity improves, specifically by around 12dB.

In RF terms 6dB improved gain in the link budget is a theoretical doubling of range, therefore 12dB improvement increases range possibilities fourfold. Link budgets can be improved by additional power, but critically the long range improvements in Bluetooth 5 do not rely on improved output power and mean up to four times more range can be achieved without a need to increase peak output power. The data throughput capacity does of course reduce when 500kbps or 125kbps modes are selected, but for many Bluetooth low energy applications which are relatively straightforward sensing and actuating applications, these data rates are perfectly adequate.

lefedor, 2018-06-27:
https://www.nordicsemi.com/eng/Products/Bluetooth-5
Have interest int this myself.
Want mention that long range is not in the nRF52832 feature list, only for the nRF52840.

hathach, 2018-07-12:
nrf52840 is coming to Adafruit store soon enough :D

mharper, 2019-01-24:
I have a relatively immediate need for this feature and have a nRF52840 Feather landing here in a few days.  @hathach if you can give me a vector or two on how to implement this, I can do it.

hathach, 2019-01-24:
This is new to me as well, you need to dive into Nordic docs and code to get it work. Please let us know what you figure out.

mharper, 2019-01-24:
Acknowledged. Will let you know what I find out.

mharper, 2019-01-29:
Based on what I've pieced together so far, the best example of how to do this is in the nRF SDK example found here:
```
examples/ble_central_and_peripheral/experimental/ble_app_att_mtu_throughput
```

Relevant struct: [`ble_gap_phys_t`](https://www.nordicsemi.com/DocLib/Content/SoftDevice_API_Doc/S140/v6-1-0/structble__gap__phys__t)
Therein lie: `tx_phys` and `rx_phys`. The value to use there for extended range appears to be [`BLE_GAP_PHY_CODED`](https://www.nordicsemi.com/DocLib/Content/SoftDevice_API_Doc/S140/v6-1-0/group_ble_gap_phys#gaca98a1c4c8c2e2bc385cd81e9109068a).
Relevant API call: [`sd_ble_gap_phy_update`](https://www.nordicsemi.com/DocLib/Content/SoftDevice_API_Doc/S140/v6-1-0/group_ble_gap_functions#gab845ddf7b56ef556ab9c2c74d5199ff8)

So in the context of `bluefruit.cpp`, I imagine the call might look like this to set the PHY:
```C++
bool AdafruitBluefruit::setPhy(int8_t phy) {
  ble_gap_phys_t phys;
  phys.tx_phys = phy;
  phys.rx_phys = phy;
  sd_ble_gap_phy_update(_conn_hdl, &phys);
}
```

Enabling extended range would then look like:
```C++
Bluefruit.setPhy(BLE_GAP_PHY_CODED);
```

You can set the PHY to any one of:
* `BLE_GAP_PHY_AUTO`
* `BLE_GAP_PHY_1MBPS`
* `BLE_GAP_PHY_2MBPS`
* `BLE_GAP_PHY_CODED` (nRF52840 only)

syed93uthman, 2019-04-11:
this feature will help in my project alot. Thanks

hathach, 2019-04-11:
@mharper thank you very much for your findings. I will do some simple tests here, but currently occupied by usb work.

jps2000, 2019-07-05:
any chance of progress here?
thanks

ladyada, 2019-07-07:
@jps2000 no ETA

mharper, 2019-07-07:
Sorry, folks. I was able to solve my problem by cranking up the tx power, so I haven't revisited this.

hathach, 2019-07-07:
@mharper you have nothing to sorry. We are busy with usb as well. Also long range need a bitbof field testing as well. We are in no hurry, still lots of other work to do.

ApoorvaSheth, 2019-11-25:
I want to test the long range of Adafruit nRF52840 express feather(BLE 5.0) board. I did the setup on arduino IDE. My question is can I perform the long range test of BLE using an arduino ide on adafruit nRF52840 express feather board?

hathach, 2019-11-25:
The feature is not yet implemented.

ApoorvaSheth, 2019-11-26:
> The feature is not yet implemented.

Not implemented on Arduino IDE or Adafruit board

ApoorvaSheth, 2019-11-27:
@hathach Is that feature is not implemented on arduino IDE or not implemented on Adafruit nRF52840 Express feather board
Please let me know

ApoorvaSheth, 2019-11-27:
I wanted to know if the long-range feature is supported in BlueFruit library for nRF52840 express.

ladyada, 2019-11-27:
@ApoorvaSheth the answer is above, if you post spam or nags again, you will get blocked

orhanyor, 2020-02-06:
this would be a good addition, i was looking how to add pa lna to 52840s to improve the range but long range support might be a better and neater solution.

path9263, 2020-04-28:
I am curious if someone can roughly outline what else is required to get this working.  I have some time to try get it going but am unsure where to start.  Thanks!

hathach, 2020-04-29:
> I am curious if someone can roughly outline what else is required to get this working. I have some time to try get it going but am unsure where to start. Thanks!

You probably need to advertise and connect using CODED PHY, check out articles in my 1st post, nordic devzone and sdk are a good source of reference.

vladkozlov69, 2020-04-29:
@path9263  you can find some ideas in my fork where I added some very basic implementation for long range, see https://github.com/vladkozlov69/Adafruit_nRF52_Arduino/pull/17/commits

path9263, 2020-05-01:
@vladkozlov69 thanks for that, very helpful.  I feel like I am pretty close but I am wondering if you overcame the problem I am having.  My central sees my peripheral advertising on Coded Phy and attempts to connect but the central fails and reports: `bool BLECentral::connect(const ble_gap_addr_t*): 78: verify failed, error = NRF_ERROR_RESOURCES`  which seems to point to setting a larger value for `_sd_cfg.central.event_len`.  Tried several values and not having any luck with that though, same error.  I see you added `uint32_t _ram_start;` to `bluefruit.h` but couldn't find it used anywhere and am wondering if that has something to do with this.  I do _not_ get an error about SoftDevice config requires more SRAM though.  Thanks!

vladkozlov69, 2020-05-01:
@path9263 I had the similar issue with NRF_ERROR_RESOURCES, and when I increased the event_len, it has gone. Usually, Coded Phy requires bigger event_len than a regular connection.

Colas-Lepoutre, 2020-06-16:
Hi,
I was wondering if there were any news about the long range feature. I basically only need to have a "setPhy" function in order to make the device advertise and connect using phy coded.
Is this something easily feasable ? Did you already get this working ?

Thanks in advance,
Colas LEPOUTRE

gernot1972, 2020-08-11:
Is there any example code for read/write characteristics with Coded PHY enabled via arduino IDE?

iamcalledusman, 2021-03-07:
Hi All !

is there any example for beacon send and receive using coded phy via arduino IDE / Platform for RAK4631 ?

Thanks In advance !

bfribush, 2022-06-21:
Has the Coded PHY feature been implemented?

naveen-79, 2025-06-03:
any update on this feature