adafruit/Adafruit_nRF52_Arduino #83: High Throughput (2 Mbps PHY) support

hathach, 2017-10-10:
**Higher data throughput**

The data rate for Bluetooth low energy has now been extended from a singular transmission speed (1Mbs) to four discrete data rate speeds as seen in the list above. The maximum speed is doubled to 2Mbs. Together with advancements introduced in Bluetooth 4.2 which allowed for Data Length Extensions (DLE) overall throughput is x5 higher than the original Bluetooth 4.0 level. Now net data throughput not including overhead such as addressing can be as high as 1.4Mbs. Data packet size, datarate and overall data throughput are illustrated in fig. 1.

closed 2019-08-16 by ladyada

hathach, 2019-08-16:
implemented #324

ApoorvaSheth, 2019-11-27:
I wanted to know if the long-range feature is supported in BlueFruit library for nRF52840 express.

PR #324 (2019-08-16): Maximize throughput
- add requestPHY() to switch to 2MB PHY, fix #83  #80
- add requestDataLengthUpdate() to increase data length
- add requestMtuExchange() to actively exchange MTU, fix #106

update throughput example, and added central_throughput example