adafruit/Adafruit_nRF52_Arduino #88: writeCallback should be called using ada_callback as well

reikjarloekl, 2017-10-24:
When reacting to a write to a characteristic, the BT stack is blocked until the action triggered by the write is finished. The mechanisms to decouple the action from the write are all already in place (--> ada_callback) but are currently not used for calling the writeCallback of a characteristic.

Would be great to at least have this as an optional behavior.

hathach, 2017-10-24:
yeah, ada_callback is introduced for that reason but later in the development. There is some left over. Can you post the exact line of code.

reikjarloekl, 2017-10-24:
[Line 330 in BLECharacteristic.cpp](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/b193c9b03be2aa7544d72a0fd029c9b3fb2cec8a/libraries/Bluefruit52Lib/src/BLECharacteristic.cpp#L330)

hathach, 2018-01-05:
requires callback signature change to use pointer to characteristic instead of reference. This will allow us   to pass it through ada callback (as middle man). Will be set to later release

closed 2019-04-04 by hathach