adafruit/Adafruit_nRF52_Arduino #91: BLECharacteristic should support long writes

reikjarloekl, 2017-10-29:
Sending long (>20 bytes) characteristics to the central works nicely already but receiving long writes from central does not currently. 

There is an example for how receiving long writes is done [here](https://github.com/NordicSemiconductor/ble_app_hrs---LongWrite).

Nordic also provides a sequency chart [here](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v3.0.0%2Fgroup___b_l_e___g_a_t_t_s___q_u_e_u_e_d___w_r_i_t_e___b_u_f___a_u_t_h___m_s_c.html).

I am willing to help implement this.

hathach, 2017-10-31:
Thanks for openning the issue, we will check it. If you like you could submit a pull request, we will revise and merge it.

jardous, 2018-08-29:
any progress on this?

hathach, 2018-08-29:
We didn't have time to check this yet. Stil focusing on the upcoming nrf52840.

hathach, 2019-02-27:
implemented by above commit

closed 2019-02-27 by hathach