adafruit/Adafruit_nRF52_Arduino #97: MAC address of ble devices storing backwards

ynohtna92, 2017-11-21:
In your example ino you have this line 
Serial.printBuffer(report->peer_addr.addr, 6, ':');

This will output the ble device address as 60:15:D8:CE:20:EF when it should be EF:20:CE:D8:15:60 instead according to the nRFConnect nordic android application.

It seems like the MAC address is being stored backwards somewhere.

hathach, 2017-11-21:
little endian is native format used by Softdevice. Question should be asked in our forums

closed 2017-11-21 by hathach

microbuilder, 2017-11-21:
As Thach mentionned, there is some ambiguity here since internally they are used one way, but conventions might be another way elsewhere. :( There are two ways to store byte records in MCUs, referred to as 'little-endian' and 'big-endian', and little-endian is used in this case for ease of use with the SD. You can of course 'flip' the byte order yourself if you want better consistency with specific applications.

hathach, 2017-11-22:
update: We also fix to print MAC address in natural order  in  all examples. Thanks