adafruit/Adafruit_nRF52_Arduino #100: BLE central client keeps disconnecting every 60 seconds

ynohtna92, 2017-11-28:
Every 60 seconds of having a client connected. It will suddenly disconnect. Is the central not responding to some sort of heart beat the is happening?

What could be causing this to occur so regularly?

Even with multiple devices connected over BLE to the central, all clients are experiencing the same issue.

hathach, 2017-11-29:
Can you post your central sketch along with peripheral sketch. We don't see this issue while testing our stock examples.

ynohtna92, 2017-11-29:
Central on Bluefruit
```#include <bluefruit.h>

char Rxchar, Rxptr, Rxenable, Cmdcomplete;
char Rxbuf[32], Cmdbuf[3], Databuf[32];
uint8_t Adbuf[6];
uint16_t Readbuf[4];

char count_wl = 0;
uint8_t whitelist[BLE_CENTRAL_MAX_CONN][6];

// Battery Service               : 0000180F-0000-1000-8000-00805F9B34FB
// Notification Source (0x000c)  : 00002902-0000-1000-8000-00805F9B34FB
// Data Source (0x000b)          : 00002A19-0000-1000-8000-00805F9B34FB

const uint8_t BMU_UUID_SERVICE[]
{
    0xFB, 0x34, 0x9B, 0x5F, 0x80, 0x00, 0x00, 0x80,
    0x00, 0x10, 0x00, 0x00, 0x0F, 0x18, 0x00, 0x00
};

const uint8_t BMU_UUID_CHR_NOTIFICATION[]
{
    0xFB, 0x34, 0x9B, 0x5F, 0x80, 0x00, 0x00, 0x80,
    0x00, 0x10, 0x00, 0x00, 0x02, 0x29, 0x00, 0x00
};

const uint8_t BMU_UUID_CHR_DATA[]
{
    0xFB, 0x34, 0x9B, 0x5F, 0x80, 0x00, 0x00, 0x80,
    0x00, 0x10, 0x00, 0x00, 0x19, 0x2A, 0x00, 0x00
};

// Struct containing peripheral info
typedef struct prph_info_t {
  prph_info_t()
  : _service(BMU_UUID_SERVICE), _notification(BMU_UUID_CHR_NOTIFICATION), _data(BMU_UUID_CHR_DATA)
  { }
  
  char name[32];
  uint8_t address[6];
  bool buckled_state;
  int battery_level;

  uint16_t conn_handle;

  BLEClientService _service;
  BLEClientCharacteristic _notification;
  BLEClientCharacteristic _data;
} prph_info_t;

prph_info_t prphs[BLE_CENTRAL_MAX_CONN];

// Software Timer for blinking the RED LED
SoftwareTimer blinkTimer;
uint8_t connection_num = 0; // for blink pattern

void setup() {
  Serial.begin(115200);

  Serial.println("Bluefruit52 Central Unit");
  Serial.println("-----------------------------------\n");

  Serial.println("Max connections: " + String(BLE_CENTRAL_MAX_CONN));

  uint8_t buf1[6] = {249, 19, 13, 65, 108, 21}; // F9:13:0D:41:6C:15
  uint8_t buf2[6] = {0, 1, 2, 3, 4, 5};
  uint8_t buf3[6] = {6, 7, 8, 9, 10, 11};
  uint8_t buf4[6] = {12, 13, 14, 15, 16, 17};

[... 1481 lines cut ...]
    timers_init();
    ble_stack_init();
	  device_manager_init(erase_bonds);
		adc_configure();
    gap_params_init();
    services_init();
    advertising_init();
    conn_params_init();
		
		

    // Start execution.
    advertising_start();
		
		//app_timer_start(m_gpio_outside_timer_id, GPIO_OUTSIDE_MEAS_INTERVAL, NULL);
		//TIMEOUT --> POWER OFF

    // Enter main loop.
    for (;;)
    {
        power_manage();
    }
}
 
```

hathach, 2017-11-29:
the code is fairly long, we won't have time to troubleshoot it.  Could you try with our stock example to see if the issue still occurred ? Otherwise it could be some logic issue in your sketch ??!!

ynohtna92, 2017-12-05:
Ok we tested it with another bluefruit sketch and it stayed connected.

We did some further investigation that let us to find out the nordic device we have will ask the central for parameters but never receive them and timeout after a minute and 5 seconds. We removed this from our embedded device and it stays connected.

Is it possible to add the parameters reply in a future version of central?

hathach, 2017-12-05:
can you provide more specific detail on the parameter and/or Nordic event name ?

ynohtna92, 2017-12-05:
I believe it relates to this.

`/**@brief Function for the GAP initialization.
 *
 * @details This function sets up all the necessary GAP (Generic Access Profile) parameters of the
 *          device including the device name, appearance, and the preferred connection parameters.
 */
static void gap_params_init(void)
{
    uint32_t                err_code;
    ble_gap_conn_params_t   gap_conn_params;
    ble_gap_conn_sec_mode_t sec_mode;

    BLE_GAP_CONN_SEC_MODE_SET_OPEN(&sec_mode);

    err_code = sd_ble_gap_device_name_set(&sec_mode,
                                          (const uint8_t *)DEVICE_NAME,
                                          strlen(DEVICE_NAME));
    APP_ERROR_CHECK(err_code);

    memset(&gap_conn_params, 0, sizeof(gap_conn_params));

    gap_conn_params.min_conn_interval = MIN_CONN_INTERVAL;
    gap_conn_params.max_conn_interval = MAX_CONN_INTERVAL;
    gap_conn_params.slave_latency     = SLAVE_LATENCY;
    gap_conn_params.conn_sup_timeout  = CONN_SUP_TIMEOUT;

    err_code = sd_ble_gap_ppcp_set(&gap_conn_params);
    APP_ERROR_CHECK(err_code);
}`

hathach, 2017-12-05:
It is possibly the inappropriate/not supported value for ppcp. If you are sure those are the one causing the issue, please open another issue with above code + value for those pcpp (MIN_CON_INTEVAL etc...). We will test with those values when having time. 

I will close this one soon enough

closed 2017-12-07 by hathach