adafruit/Adafruit_nRF52_Arduino #103: [Q] Backwards compatibility with standalone nRF51?

floe, 2017-12-07:
As far as I could tell, the core and libraries here should also be able to run on a standalone nRF51 board, correct? (as long as the lower RAM/memory limits are observed.)
Note: I know that this does not apply to the nRF51-based Adafruit boards such as Feathers, but something like a BBC micro:bit should be compatible?

microbuilder, 2017-12-09:
This is theoretically possible, I suspect, but not something we would do ourselves and wouldn't be a trivial effort. The nRF51 and nRF52 products we sell have two entirely different design approaches, and we think the choice of something AT based (nRF51) and direct API access based (nRF52) makes sense to maintain at the moment.

closed 2017-12-09 by microbuilder

floe, 2017-12-11:
Thanks - the main reason why I asked is that there's now basically two competing APIs for supporting BLE on the nRF5x series with Arduino. OTOH, there's https://github.com/sandeepmistry/arduino-BLEPeripheral, and on the other hand, there's https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/Bluefruit52Lib.

Could you comment briefly on why you think this will take significant effort? AFAICT an additional board definition might be sufficient, as the newer Nordic SDKs are backwards-compatible?

mchinand, 2017-12-11:
The nRF52-based boards are designed differently from the nRF51-based boards. The latter have a 2nd microcontroller that is the 'main' controller on the board and is connected to all the pins on the board. On these boards, the Nordic chip is only providing the Bluetooth functionality. The nRF52-based boards only have the one Nordic microcontroller to provide all functionality.

floe, 2017-12-11:
@mchinand yes, thanks, I was aware of that (see also initial post). My question relates to the very popular boards like the BBC micro:bit which are based on the nRF51 alone, and which are also supported by the Arduino IDE thanks to @sandeepmistry.

ladyada, 2017-12-11:
@floe nRF51 is deprecated, and very underpowered at only 16 mhz, so our focus is on the nRF52 which can chew gum and walk at the same time.

what speficially are you trying to do?

floe, 2017-12-12:
I have a bunch of micro:bit boards (which are based on the nRF51), and my students have a good workflow with the Arduino IDE at the moment. However, all BTLE-related development currently happens using https://github.com/floe/arduino-BLEPeripheral (patched from sandeepmistry), and this library starts to show its age IMHO. So ideally, I'd like to switch to using the Adafruit library, which gives me an upgrade path to nRF52 boards, but use it on my existing nRF51 boards for now.

ladyada, 2017-12-12:
@floe , if you are able to port this library to nRF51 without affecting nRF52 functionality we'll be happy to look at a pull request! we have no plans to do the work ourselves as we are already very busy supporting our own products :)

floe, 2017-12-12:
@ladyada completely understandable :-) I'll see if my original idea of just adding a tweaked board definition for the micro:bit is feasible - PR hopefully to follow soon.

microbuilder, 2017-12-14:
You'll probably need to keep the serial bootloader in mind as well, although the micro:bit boards do have a CMSIS-DAP debugger on board which can be used from the Arduino IDE to flash as well, though the BSP might (or might not!) need some tweaks for this, I've never tested.