adafruit/Adafruit_nRF52_Arduino #109: FastLED port

hathach, 2017-12-16:
https://forums.adafruit.com/viewtopic.php?f=57&t=124748

https://github.com/FastLED/FastLED/issues/476

henrygab, 2019-04-11:
I'm working on a full port now.  I did use pcaddict's branch, also with APA102.  However, it's not quite as stable as it should be.   Esp. needs the ability to set H0H1 for high-speed SPI, and support for boards where the Arduino pin numbers don't match the board pin numbers....

hathach, 2019-04-13:
Thanks @henrygab , please keep us update and submit PR if it is ready.

henrygab, 2019-04-16:
Update:
FYI, there are three parts to getting FastLED support.
1. FastPIN -- allows toggling individual pins very rapidly.  FastPIN is sufficient to control APA102 (dotstar) or other SPI LEDs using using software SPI.
2. FastSPI -- Enables hardware-assisted SPI, improving performance.
3. Clockless -- Enables neopixel and other implied-clock chips.

A **[PR for Part 1](https://github.com/FastLED/FastLED/pull/774)** (FastPin) is now submitted for FastLED's maintainer (focalintent) to code review / consider.  See https://github.com/FastLED/FastLED/pull/774.

I am moving on to Part 2 (FastSPI), which I already have in mind how to get working.

I am unlikely to complete Part 3 (clockless), but FastLED's maintainer (focalintent) may try to get and validate that one.

hathach, 2019-04-19:
superb !!!

henrygab, 2019-04-24:
An updated **[PR for Parts 1 and 2](https://github.com/FastLED/FastLED/pull/779)** (FastPin and FastSPI) has now been submitted for FastLED's maintainer (focalintent) to code review / consider.  See https://github.com/FastLED/FastLED/pull/779.

### Good enough for SPI-based LEDs
This works for APA102 (DotStar), as an example.

### Missing for full support
All that is needed is **clockless**.  Unfortunately, that's beyond my abilities at the moment.  If you have the ability, see the comment I added to the above PR, for initial notes on the information I already discovered that may be useful.

henrygab, 2019-04-29:
For those reaching this item, wondering how to use the nRF52 library with SPI chipsets, even before it's integrated into the official FastLED libraries, here's one method, based generally on https://www.arduino.cc/en/Guide/Libraries#toc5, but using `git` instead of downloading a ZIP file.

#### Find your sketchbook location
1. Start Arduino IDE
2. Menu Option **`File->Preferences`**
3. Note the directory listed as you **`Sketchbook location`**
_This example will use a sketchbook location of `C:\Users\ExampleUser\Documents\Arduino` … replace with your path when you see this path below._

#### Open a command-line and go to the libraries folder
For Windows:
1. Click `Start->Run`, type **`cmd.exe`**, and press enter
2. type **`cd /d c:\Users\ExampleUser\Documents\Arduino`** to go to sketchbook folder (replace with your sketchbook path from above)
3. type **`cd /d Libraries`** to enter the libraries folder
4. keep this window open for next steps...

#### Remove any prior version of FastLED
For Windows, using GUI:
5. type **'start .'** … a file explorer window should open showing the libraries folder
6. If the folder 'FastLED' exists, delete it (and all files inside)
7. Close the file exporer window (keep the cmd.exe window)

#### Use GIT to download the nRF52/SPI enabled FastLED libraries
In the cmd.exe window....
8. type **`git clone https://github.com/henrygab/fastled.git`** and press enter
9. type **`git checkout nrfx`** and press enter (_this will obtain the branch with nrf52 support_)

That's all it takes.

henrygab, 2019-05-22:
PR for a first version with support for clockless:
https://github.com/FastLED/FastLED/pull/779

Only update to the above instructions, is to change the branch name from `nrfx` to `nrf52`.
For example, replace line:
`git checkout nrfx`
with
`git checkout nrf52`

Works as-is up with strings of clockless LEDs up to 144 LEDs long.  Simply modify the line (currently at [clockless_arm_nrf52.h line 10](https://github.com/henrygab/FastLED/blob/bc62bd20eb41f562f25b9169746e214fa52d208e/platforms/arm/nrf52/clockless_arm_nrf52.h#L10)
`#define FASTLED_NRF52_MAXIMUM_PIXELS_PER_STRING 144`

Allocations are done at compile-time, so you'll know if you have enough RAM when you compile.

henrygab, 2019-06-07:
PR https://github.com/FastLED/FastLED/pull/802 was accepted a few hours ago.
Soon afterwards, FastLED revision was [updated](https://github.com/FastLED/FastLED/blob/0cea6c8863f8ef9978923377d476d3510c5830b7/library.json#L25) to version 3.0002.0007, including explicitly listing support for the [nordicnrf52](https://github.com/FastLED/FastLED/blob/0cea6c8863f8ef9978923377d476d3510c5830b7/library.json#L25).

Accordingly, I believe this issue can now be resolved.  👍

henrygab, 2019-06-07:
PR https://github.com/FastLED/FastLED/pull/802 was accepted a few hours ago.
Soon afterwards, FastLED revision was [updated](https://github.com/FastLED/FastLED/blob/0cea6c8863f8ef9978923377d476d3510c5830b7/library.json#L25) to version 3.0002.0007, including explicitly listing support for the [nordicnrf52](https://github.com/FastLED/FastLED/blob/0cea6c8863f8ef9978923377d476d3510c5830b7/library.json#L25).

@hathach I believe this issue can now be resolved.  :tada:

hathach, 2019-06-07:
Superb!!! Thanks @henrygab, though I have no idea how to use this yet (wil try later). But it is great addition. @ladyada do you happen to know FastLED and/or used it previously on samd 😊😊

ladyada, 2019-06-07:
it definitely works on SAMD because we added support for our boards a while ago. you can use it with any WS2812-compatible LEDs to test

hathach, 2019-06-11:
thanks @henrygab  again for the excellent works :)

closed 2019-06-11 by hathach