adafruit/Adafruit_nRF52_Arduino #113: BLEClientCharacteristic events for "indicate" notifications

savageautomate, 2018-01-12:
How does one receive events for "**indicate**" notifications on a CHARACTERISTIC?
I tried using the `BLEClientCharacteristic::setNotifyCallback() `method, but I'm not seeing any callback invocations for indicated events.  I am seeing the following debug statements printed:

```
[BLE   ] BLE_EVT_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_HVX : Conn Handle = 0
```

In looking at the code, I see a placeholder in BLEClientCharacteristic.h for:

`typedef void (*indicate_cb_t) (BLEClientCharacteristic& chr, uint8_t* data, uint16_t len);`

But there is no method to establish the callback function pointer such as:
`void     setIndicateCallback(indicate_cb_t fp);`

Also in `BLEClientCharacteristic.cpp::_eventHandler(ble_evt_t* evt)`, there is no `BLE_GATTC_EVT_HVX` type handler for `BLE_GATT_HVX_INDICATION`, only `BLE_GATT_HVX_NOTIFICATION`.   
(https://devzone.nordicsemi.com/documentation/nrf51/4.4.0/html/group___b_l_e___g_a_t_t___h_v_x___t_y_p_e_s.html)

I'm new to this, so my assumptions above may be wrong, but I'm just struggling to try and get events for a characteristic on an embedded device that only uses INDICATE and not NOTIFY.  

Thanks!

hathach, 2018-01-13:
We are glad that you ask, the indicate haven't used so far. So we forgot to add it to the API. We will include it (first in the develop branch), then in the next release.

hathach, 2018-01-13:
PS: We are planning to change the signature of callback for better deferred with adacallback as well.

savageautomate, 2018-01-15:
Great, looking forward to testing the fix.  Thanks!

hathach, 2018-01-18:
we just pushed the new code to branch 0.8.x that implement the indicate callback. 
If you could test it out, we will include it in the next release.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/0.8.x/libraries/Bluefruit52Lib/src/BLEClientCharacteristic.h#L104

Note: the callback signature is changed from using Reference to pointer. This allow easy integration to ada callback. Which is a separate thread used for all the callback. using setNotifyCallback()/setIndicateCallback(), you could choose whether callback should be called in callback thread or main ble thread (blocking or event). 

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/0.8.x/libraries/Bluefruit52Lib/src/BLEClientCharacteristic.h#L50

hathach, 2018-01-18:
There are cases where you want a callback is called in main ble thread. E.g you are in characteristic A's callback (in callback thread) and waiting for another event of characteristic B's callback. Therefore A's callback and B's callback must be on separate threads (ble & callback thread). Otherwise it will go into deadlock.

savageautomate, 2018-01-18:
I spent a few minutes today setting up my environment for testing the development brach (0.8.x) locally.  On a first cursory attempt ... I am now receiving characteristic indication events issued by a BLE (hardware) device. 

I'll spend a little more time tomorrow digging in a bit further .. but initial test is all good!

Thank You

hathach, 2018-01-19:
Thanks, we are looking forward to hear from your testing. Once it is done, we will release it as 0.8.2 . Bug discovered later will fire as separate bug/issue.

hathach, 2018-01-22:
we will release the code as 0.8.2 and close this issue, please file an different issue as bug if you have any troubles with this.

closed 2018-01-22 by hathach

savageautomate, 2018-01-22:
In my limited testing over the weekend it all seemed to work fine.  
Thanks again for the quick fix.

hathach, 2018-01-23:
@savageautomate thanks for your confirmation. It is good enough to roll out for other folks to test with :)