adafruit/Adafruit_nRF52_Arduino #116: Central example request - HID keyboard

ladyada, 2018-02-05:
not an immediate need, but an interesting demo would be if the nRF52 was a *central* to an HID BLE keyboard. hardest part is finding a BLE keyboard (not BT classic!)

hathach, 2018-02-12:
just tried to implement it, parsing characteristic and getting report is easy enough. Most keyboards requires pairing/bonding, which is unfortunately not yet implemented on Central (only Peripherals). I will dig up some docs and try to wrap this up.

hathach, 2018-02-23:
just added the central hid example. i don't have a BLE keyboard to test with but it works well with current prph keyboard running on another feather nrf52. It may be buggy a bit but work as proof of concept.  
 
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/0.8.x/libraries/Bluefruit52Lib/examples/Central/central_hid/central_hid.ino

hathach, 2018-02-26:
If anyone that has consumer ble keyboard to test with, please let us know if that works out

microbuilder, 2018-02-26:
@hathach I can test this out.

One request to close this though is to add a function to map raw HID keycodes to a country-specific keyboard layout.

For example, on a French keyboard, the top-left keys are `azerty`, so the `q` keycode needs to be mapped to `a`, and the `w` keyboard needs to be mapped to `z`, etc. We should have a `.mapKeycode` type function that accepts a lookup table, and returns the country-specific values so that this can be used with any keyboard with a bit of manual labour on behalf of customers (providing a new keymap lookup table).

microbuilder, 2018-02-26:
Additional note: The lookup function needs to support multiple keys to know if you are generating special characters, upper-case, etc.

microbuilder, 2018-02-26:
The demo looks great and is easy to understand with the callbacks. Great job on this!

wwwoholic, 2018-11-17:
The comment in central_hid example "A additional bluefruit that has bleuart as peripheral is required for the demo" doesn't seem to be right. Also, the notifications for both keyboard and mouse are enabled, but only keyboard reports are handled.

hathach, 2018-11-19:
thanks We will update the comment and rename the example to central_hid_keyboard, the mouse API is similar. We would add it later when having more time