adafruit/Adafruit_nRF52_Arduino #117: BLEUart doesn't work with S132 5.1.0

ddcc, 2018-02-12:
I noticed that support for the Nordic S132 5.1.0 SoftDevice was added recently, so I upgraded by burning the new bootloader using the Bootloader DFU, and re-flashing my sketch.

But, when I test with both the `nRF Connect` and `nRF UART v2.0` Android applications, the BLEuart service doesn't seem to receive data anymore, as `bleuart.available()` never returns non-zero. I double checked with the example `bleuart.ino` sketch, and experienced the same behavior. When I downgraded back to S132 2.0.1, it works fine, so this seems to be a regression.

One thing that I did notice, is that when reading the `Client Characteristic Configuration` field (UUID 0x2902) with `nRF Connect` on S132 5.1.0, the response significantly exceeds 16 bits and seems to be invalid.

Attached are the two different sanitized logs: [s132_v201.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/1715445/s132_v201.txt) [s132_v510.txt](https://github.com/adafruit/Adafruit_nRF52_Arduino/files/1715446/s132_v510.txt). Notice that the RX and Notify work fine with 2.0.1, but nothing is printed after the connection on 5.1.0.

hathach, 2018-02-12:
Could you tell which BSP you are using, and did our Bluefruit app work with the bleuart sketch

ddcc, 2018-02-12:
Both were with 0.8.2. On S132 2.0.1, Bluefruit app UART worked, but not with S132 5.0.1.

hathach, 2018-02-12:
Ah Thanks, can you enable Debug level = 2 and paste your serial log here.

ddcc, 2018-02-12:
Yeah, see the attached logs above.

hathach, 2018-02-12:
I just double checked, It works OK with my iphone6 and Nexus 5. Hmmm. Look like you are paired with the feather. It could make your phone "remember/cache" some setting. Could you try to unpair them, turn off/on bluetooth and try again ?

ddcc, 2018-02-12:
Thanks for the suggestion! Turns it wasn't just paired, but also bonded. After deleting the bond information and retrying, it works now.

closed 2018-02-12 by ddcc