adafruit/Adafruit_nRF52_Arduino #120: Nffs file system:only examples works!!

pbecchi, 2018-02-17:
I am disappointed ...i try to use Nffs file system ....nothing seams to work! (except examples)!

hathach, 2018-02-17:
If you have problems using nffs or any repo code, please post it in the support forums. There is support team that will help you out. Github is mainly used for repo development such as filing bugs and features

closed 2018-02-17 by hathach

pbecchi, 2018-02-17:
Do you think there is no bug?
Are you sure that all Nffs features works?

microbuilder, 2018-02-17:
We're /unaware/ of any bugs, but if you can produce code that clearly/repeatably demonstrates a bug we'll be happy to address it, and Github is the right place for that. As for 'How do I use this', that should happen on the Adafruit Support Forums, where we have support staff that actively monitors and escalates issues. The example code that we provide does work, and NFFS is used internally in the firmware and also works, so we haven't seen 'bugs', but that doesn't mean there aren't anyway, but we need something concrete to reproduce them first.

pbecchi, 2018-02-17:
Hi thanks for your answer! I am porting a Esp8266 code to nRF52, the only difficulties I have found are relative to Nffs!
The example run but show only 2 very simple methods read(bytes, bytelength), write (bytes, bytelength)
Since the code was hanging or giving unpredictable results, I have started modifying readwrite.ino example testing use of various methods. Both write and print method if called more that 2 or 3 times hang the nRF52. Indexing (file.seek() ) and writing over an existing file do not change the content on flash.
The only operation I can do successfully is to write one buffer and close the file.Then I am able to read what I have written!
I am solving my needs using a large buffer that I read, modify and overwrite .......I hope there will be some time a better way to use a file system on nRf52!

hathach, 2018-02-19:
That is lots of work and too wide for us to investigate. It could be other part of the cose that cause the issue as well. If possible please share your simplest sketch that contains only the nffs code that has bug. The more simple the test sketch, the easier for us troubleshoot it. 

When you do please open a seperate issue with more specific information

pbecchi, 2018-02-24:
Sorry for this late answer, but I have been working hard with Nrf52 code, trying to overcame all obstacles....
Finally the code compile and run but....
My main concern now is to reduce an abnormally high power consumption with an average of 6 to 8 mA.
I have to reduce that to less then 50 uA so 100 time less! I think it is feasible and I have started experimenting on some very simple test code going down to 10 uA.....
I would like to share all my findings....you tell me if you prefer that this will be done here or in your support forum.

hathach, 2018-02-25:
That is great, we would like to know what you have done to reduce power. There is an ongoing issue with power consumption here #51 . Maybe you could share your finding there. We are still working on it.