adafruit/Adafruit_nRF52_Arduino #127: J-Link Bootloader

Douglas-5DT, 2018-03-01:
Hi, 
I am using the Adafruit Bluefruit nRF52 feather with Adafruit nRF52 v0.8.2 installed in Arduino 1.8.5 IDE.
I'm trying to use a J-link to burn the bootloader but I get the following error: 

Arduino: 1.8.5 (Windows 10), Board: "Adafruit Bluefruit nRF52 Feather, 5.1.0 dual, S132 5.1.0, Level 0 (Release)"

java.io.IOException: Cannot run program "nrfjprog.exe": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:412)
	at processing.app.Editor.lambda$handleBurnBootloader$14(Editor.java:2519)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 5 more
Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I have the following settings under tools:
Board: Adafruit Bluefruit nRF52 Feather
Bootloader: 5.1.0 dual, S132 5.1.0
Programmer: J-link for Bluefruit nRF52

I can manually burn the bootloader when I use command line, but not when using the Arduino IDE. The J-Link also works when I use Adafruit nRF52 v0.7.5 in the IDE

hathach, 2018-03-01:
please post on the forum for support, github is used for code development.

closed 2018-03-01 by hathach

microbuilder, 2018-03-01:
Hi, as mentioned this should probably go in the forums (since other people will be more likely to find the thread, compared to here on Github where once something gets closed it tends to disappear into a vacuum). Feel free to post the question there, but also make sure `nrfjprog` is available in your system path, since it looks like it isn't being found and it isn't a standard part of the nRF52 BSP from us so it needs to be installed manually and made available in the `PATH` variable by you since this is very much an edge case in terms of board usage.