adafruit/Adafruit_nRF52_Arduino #128: NFFS API is "eating" native errors

xkortex, 2018-03-13:
Take, for instance, `Nffs.mkdir` vs `fs_mkdir`:

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/a86fd9f36c88db96f676cead1e836377a37c7b05/libraries/nffs/src/Nffs.cpp#L104

It's only checking if the result is `FS_EOK`. However, there are a bunch of possible return codes:
```
#define FS_EOK          0  /* Success */
#define FS_ECORRUPT     1  /* File system corrupt */
#define FS_EHW          2  /* Error accessing storage medium */
#define FS_EOFFSET      3  /* Invalid offset */
#define FS_EINVAL       4  /* Invalid argument */
...
#define FS_EEXIST       11  /* File or directory already exists */
#define FS_EACCESS      12  /* Operation prohibited by file open mode */
#define FS_EUNINIT      13  /* File system not initialized */
```
I'm trying to debug issues with using the Nffs module, and just getting a worked/failed result is making things tricky. Before I go ripping things up and trying to put `Serial.print`s in under-the-hood libraries, is there any way to get at these codes, without resorting to using the `fs` calls directly?

I don't understand why this module is choosing to throw away useful information.

hathach, 2018-11-06:
notes: nffs will be replace by arm littlefs https://github.com/ARMmbed/littlefs in the next major release, we will close this issue later on

hathach, 2018-11-08:
replaced by littleFS https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/FileSystem/src

closed 2018-11-08 by hathach