adafruit/Adafruit_nRF52_Arduino #129: NFFS Erase example is failing in project code, works fine on its own.

xkortex, 2018-03-13:
I'm working off of the [EraseNffs example](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/develop/libraries/nffs/examples/EraseNffs/EraseNffs.ino) here. This works just fine on its own. 

However, I'm trying to include this functionality as part of my project, and it is not working correctly when included in the larger codebase (proprietary, but I'll try to share the critical functionality)

Here's my serial output:

```
bool ApacheNffs::begin(): 79: No FS detected, format 
Bluefruit.begin()
Nffs.begin()
Starting advert service
Advert service started
Initiating factory reset!
root
|_ adafruit/
|  |_ bond/
|_ lost+found/

...
Nffs Erasing Example

Erasing 0x06D000 ... OK
Erasing 0x06E000 ... Failed
Exit due to ERRORs

```

```cpp
void bluefruit_init(void) {
    Bluefruit.begin();    // Bluefruit module must be initialized for Nffs to work
    // Since Bluefruit's SOC event handling task is required for flash operation
    io.printf("Bluefruit.begin()\n");
    Nffs.begin();// Initialize Nffs for persistence
    io.printf("Nffs.begin()\n");

    Bluefruit.setTxPower(4); // Set max power. Accepted values are: -40, -30, -20, -16, -12, -8, -4, 0, 4
    Bluefruit.setName(BLE_DEV_NAME);

    bleuart.begin();        // Configure and start the BLE Uart service
    startAdv();             // Roll the advertisement right into the initialization.
}
```

This is straight from the bluefruit examples:

```cpp
void startAdv(void)
{
    io.printf("Starting advert service\n");
    // Advertising packet
    Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
    Bluefruit.Advertising.addTxPower();

    // Include the BLE UART (AKA 'NUS') 128-bit UUID
    Bluefruit.Advertising.addService(bleuart);

    // Secondary Scan Response packet (optional)
    // Since there is no room for 'Name' in Advertising packet
    Bluefruit.ScanResponse.addName();
    Bluefruit.Advertising.restartOnDisconnect(true);
    Bluefruit.Advertising.setInterval(32, 244);    // in unit of 0.625 ms
    Bluefruit.Advertising.setFastTimeout(30);      // number of seconds in fast mode
    Bluefruit.Advertising.start(0);                // 0 = Don't stop advertising after n seconds
    io.printf("Advert service started\n" );
    //io.printf((int) &bleuart );
}
```

Right now I'm calling 

```cpp
void setup() {
    setup_pins();
    io.setTimeout(50); ///< something to help with the buffer pileup that occurs when reading too fast
    io.begin(BAUD); 
    bluefruit_init();
    erase_nffs();
    ...
}
```
I notice that if I knock out the `startAdv()` call in the init, then I'm able to erase all of the memory. However, this is really problematic in the long run, as I want a `factory_reset()` which can be called via command over BLE Uart.

hathach, 2018-03-31:
Sorry for letting you wait, but we are focusing on the nrf52840 release. So these not-so-easy to fix bug will need to wait until the new board come out.

hathach, 2018-11-06:
notes: nffs will be replace by arm littlefs https://github.com/ARMmbed/littlefs in the next major release, we will close this issue later on

hathach, 2018-11-08:
replaced by littleFS https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/libraries/FileSystem/src

closed 2018-11-08 by hathach