adafruit/Adafruit_nRF52_Arduino #133: Upload via programmer (JLink)

jeremypoulter, 2018-03-24:
Hi,

When trying to upload a sketch using the upload via programmer option I get the error `Error while uploading: missing 'program.params.verbose' configuration parameter`. Is it possible to support uploading via the JLink?

Cheers,

Jeremy

hathach, 2018-03-31:
Not possible now, but we will include this in future release.

jeremypoulter, 2018-03-31:
That is good to know. If you can give some hits as to what needs doing I can take a look. 

I did do some experiments with the command line  tools and as far as I could tell it was more than just flashing the relevant bit of flash. I presume there is something that needs to be set to tell the Bootloader about the new App?

hathach, 2018-03-31:
you will need to update the checksum of application hex, which bootloader uses to check application integrity. More details is mentioned in support forum. I kind of forgot it now.

hathach, 2018-04-08:
Thanks for the link, don't worry about technical issue. We are well aware of how-to, it just requires a bit more effort to put them together under Arduino with friendly user interface.

jeremypoulter, 2018-04-15:
Any chance you can give me the command(s) to generate the signature hex file?

hathach, 2018-04-16:
the signature crc is generated and packaged by nrfutil command. Please checkout the arduino compile output log for details. We have an WIP tool for that written quite some time ago

https://github.com/adafruit/Adafruit_nRF52_Arduino/tree/master/tools/pynrfbintool

It may or may not work, you have to dig into the code and try it out.

jeremypoulter, 2018-04-17:
Just for (mainly my) reference this looks to work on the command line:

```
arm-none-eabi-objcopy -O binary firmware.elf firmware.bin
python pynrfbintool.py firmware.bin
JLinkExe -device nRF52832_xxAA -speed 4000 -if swd -autoconnect 1 -CommanderScript upload.jlink
```

where `upload.jlink` contains 

```
h
loadbin firmware.bin,0x23000
loadbin firmware_signature.bin,0x7F000
r
q
```

jeremypoulter, 2019-01-07:
Just been coming back to this after updating to the new 0.9.2 BSP. Has the firmware and/or signature location changed with the new bootloader?

hathach, 2019-01-10:
@jeremypoulter just added jlink as programmer, you could **shift+upload** to use it. However, it requires 0x0001 to be written to BOOTLOADER_SETTING_ADDRESS (if it is not previously written). I could get it done in one single command with Arduino. `& and && or |` doesn't work at all. 

Workaround for this would be
1. Manullay run nrfjprog to write **0x00000001** to bootloader addres (0x7F000 for 832 and 0xFF000 for 840)
2. Did at least one DFU upload with serial (it will write value 0x01 to setting, crc is disabled by default now).

Afterwards, any jlink program should work ( since bootlaoder setting will never get touched again, unless you upgrade bootloader).

closed 2019-01-21 by hathach

christophepersoz, 2019-02-01:
@jeremypoulter, I'm just curious about your investigation on burning your firmware with JLink. Did you find a way to make it works?

Thanks

hathach, 2019-02-01:
Hold shift and click upload, though you need to do at least one time serial or disbale crc by writing to flash address ( check my above post)

christophepersoz, 2019-02-01:
Thanks @hathach, I wasn't sure after reading : **_hathach removed this from Done in Next Release 11 days ago_** above.

hathach, 2019-02-01:
It is implemented, mentioned in changelog.md

christophepersoz, 2019-02-01:
Ok. How to figure it out when not using Arduino IDE?

hathach, 2019-02-01:
You could enable Arduino verbose mode and mimic program command

TerryGeng, 2020-08-06:
Just add some details about doing so with openOCD.

After building Adafruit's bootloader https://github.com/adafruit/Adafruit_nRF52_Bootloader with 
- `make BOARD=feather_nrf52832`, (in my case, I use nrf52832)

flash it 
- with `nrfjprog` by running `make BOARD=feather_nrf52832 flash` or
- with openOCD, running `program feather_nrf52832_bootloader-_s132_6.1.1.hex verify`, where `feather_nrf52832_bootloader-_s132_6.1.1.hex` is the compiled hex file found inside `_build` folder.

So now the bootloader and softdevice have been flashed. Then we have to deal with the bootloader settings. The bootloader setting address can be found inside `linker/nrf52.ld`. It differs from board to board, so you need to choose the right `.ld` file.

It states `BOOTLOADER_SETTINGS (rw) : ORIGIN = 0x7F000, LENGTH = 0x1000`, which tells me the bootloader setting starts from 0x7F000.

The bootloader settings struct is defined in https://github.com/adafruit/Adafruit_nRF52_Bootloader/blob/master/lib/sdk11/components/libraries/bootloader_dfu/bootloader_types.h#L45-L55. I quote here
```C
typedef struct
{
    uint16_t bank_0;          /**< Variable to store if bank 0 contains a valid application. */
    uint16_t bank_0_crc;      /**< If bank is valid, this field will contain a valid CRC of the total image. */
    uint16_t bank_1;          /**< Variable to store if bank 1 has been erased/prepared for new image. Bank 1 is only used in Banked Update scenario. */
    uint32_t bank_0_size;     /**< Size of active image in bank0 if present, otherwise 0. */
    uint32_t sd_image_size;   /**< Size of SoftDevice image in bank0 if bank_0 code is BANK_VALID_SD. */
    uint32_t bl_image_size;   /**< Size of Bootloader image in bank0 if bank_0 code is BANK_VALID_SD. */
    uint32_t app_image_size;  /**< Size of Application image in bank0 if bank_0 code is BANK_VALID_SD. */
    uint32_t sd_image_start;  /**< Location in flash where SoftDevice image is stored for SoftDevice update. */
} bootloader_settings_t;
```

The first 16-bit is a flag used to indicate bank 0 stores a valid application (BTW bank 0 starts from 0x26000). The second 16-bit is CRC. Write zero to it means CRC check is disabled. What we are going to do now is set these two fields to be 1 and 0:

- `flash fillh 0x7f000 0001 1` to deal with the first field
- `flash fillh 0x7f002 0000 1`, the second field

or in one line,
- `flash fillw 0x7f000 00000001 1` (the position is reversed to match the endian)

Now you can check with `flash mdh 0x7f000` and `flash mdh 0x7f002` to see the value just wrote.
```
> flash mdh 0x7F000
0x0007f000: 0001
> flash mdh 0x7F002
0x0007f002: 0000
```

The final step is to flash your own application, 
- `program [your_hex_file] verify`
- `reset`

That's it. Enjoy.

ramsie838, 2021-07-16:
I have an ACN52840 module with an nRF52840 chip. The module has a USB interface. If I burn the Adafruit nRF52840 express bootloader on the ACN52840 module via a Segger J-link and then upload an Arduino sketch via USB, everything works fine.

I now want to program the ACN52840 exclusively via the Segger J-link. Specifically, this means that I want to transfer both the bootloader and the Arduino sketch via J-Link.

My tests have shown that you cannot easily transfer the arduino sketch as a hex file to the ACN52840. After that the program does not work. I used the nRF programmer software to transfer the hex file.

What do I have to do to get my program code to work as a hex file on the nRF52840?

Girum, 2023-06-06:
> 

YOU ARE A GOD!

I spend the whole day trying to flash... Then trying to programming. Using the arduino-nrf5, a cannot use the Adafruit libs. But flasing the feather hex, and the memory set (using an generic nrf52 Holyiot board), i can flash my arduino hex file with adafruit lib and it works.

YOU SAVED MY LIFE

Girum, 2023-06-06:
> Just add some details about doing so with openOCD.
>
> After building Adafruit's bootloader https://github.com/adafruit/Adafruit_nRF52_Bootloader with
>
> * `make BOARD=feather_nrf52832`, (in my case, I [...]

YOU ARE A GOD!

I spend the whole day trying to flash... Then trying to programming. Using the arduino-nrf5, a cannot use the Adafruit libs. But flasing the feather hex, and the memory set (using an generic nrf52 Holyiot board), i can flash my arduino hex file with adafruit lib and it works.

YOU SAVED MY LIFE