adafruit/Adafruit_nRF52_Arduino #138: [Enhancement]Bonding/Pairing with multiple peers

samveen, 2018-04-06:
Background:
I am working on a BT Keyboard, where I want to use the same keyboard with multiple devices, without repeatedly pairing against all 3 of my devices.

What I plan:
My plan is to have implement a 4-way selection switch, to decide on an active bond slot, such that any action such as pairing/un-pairing only affects the selected slot. Once the KB is paired with a device, when I need to pair with a new device, I select an new pairing slot with the selection switch, and bond with the new device. Same applied for any new device I want to pair with. This is **not** multiple concurrent pairing. It is multiple pairing with the ability to select the active pairing.

What I want to do:
I want to enhance the bonding code to allow for such an indexed keypair list, Basically instead of a single set of keys, I maintain an indexed list them, one per paired device, and allow selecting the active index. The actions without any index fall back to using index 0, keeping the older interfaces stable.

Comments on feasibility requested. I can put in an MR in case such an enhancement is acceptable.

hathach, 2018-04-06:
It should already handle multiple bonding/pairing. Could you describe your set up, and which device that bonded with the device ?

hathach, 2018-05-08:
close due to lack of user exaplanation

closed 2018-05-08 by hathach