adafruit/Adafruit_nRF52_Arduino #148: Please add example code for a custom service with a characteristic write property, or help me get mine working

ChristopherJoonMiller, 2018-05-01:
I have been trying to create my own custom service
```
BLEService remoteCommands = BLEService(0x1234);
BLECharacteristic commandCode = BLECharacteristic(0x2345);
```
I am following the example here: https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/blecharacteristic

As well as browsing the existing examples and src code.

I followed the .begin() strategy as described in the guide.
```
  remoteCommands.begin();

  commandCode.setProperties(CHR_PROPS_WRITE);
  commandCode.setPermission(SECMODE_OPEN, SECMODE_NO_ACCESS);
  commandCode.setFixedLen(1);
  commandCode.setWriteCallback(command_callback);

  commandCode.begin();
```

but I'm still not able to see the 0x1234 Service or the 0x2345 Characteristic to write to in my "BLE Scanner" app. I previously set this same concept up in the nrf51 code base and was able to see my service/characteristic that way? No luck using the nrf52 lib so far though.

I am also unable to see the custom HRM Services/Characteristics, when loading that example. Any reason why I can't see, read, or write to any custom services?

hathach, 2018-05-02:
There is plenty of example for write char. You could check out the BLEUart service implementation for example. For support and how to please post on our support forum, there is a team there to help you out. Github is only for bug and code development.

closed 2018-05-02 by hathach

JontyMC, 2018-06-21:
I would appreciate this too, can't find any examples that use CHR_PROPS_WRITE and setWriteCallback

JontyMC, 2018-06-21:
Found examples via github search https://github.com/search?q=CHR_PROPS_WRITE&type=Code

ametedinov, 2020-03-26:
I am trying to do the same thing too, but and still there is no examples on the Adafruit website or here after two years from the original post. All I want is to be able to control an LED using web-bluetooth. I want to set up a custom service and charactersistis and NOT have to use the BLE UART. Please an example or an explanation of how to handle the value in a write characteristic.

RobertMcReed, 2020-04-10:
@ametedinov there is a good example included with the Bluefruit library that you can follow.

Take a look at the _nrf_blinky_ example in the _peripherals_ section. I was also not sure how to get started with this and found it very helpful. It shows you how to setup the write callback, and how to use the received value to set the onboard LED.

Hopefully you will find it helpful. Here is a [direct link](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/2d8eb2b14a096d36003afeb9cfffaf959a7291b9/libraries/Bluefruit52Lib/examples/Peripheral/nrf_blinky/nrf_blinky.ino).