adafruit/Adafruit_nRF52_Arduino #150: Crash BLE in Central mode

antonioriccelli, 2018-05-06:
Hi
When Connect and disconnect frequently one BlueFruit Feather in Peripheral and one BlueFruit Feather in Central mode this ultime got to CRASH and must be reboot from reset button for reconnect with Peripheral

antonioriccelli, 2018-05-08:
> A: adafruit/Adafruit_nRF52_Arduino
>
>
>
> Which sketches did you use for both the Central and Prph. To rule out sketch's own issue, can you
>
> 1.	Use the stock example cental_bleuart and bleuart for both feathers
> 2.	describe specifically what you did for us to reproduce the issue. Simply power on/off the prph for dozens of time would be enough ???!!!

Yes

I test with 2 identicall feather BLE and example of arduino cental_bleuart and bleuart for both feathers

If they are connect and disconnect BLE wireless connections, the BLE Central go to Crash and do not receive anymore….

 

Antonio Riccelli

Mobile phone +39 333 89 14 105

Office phone +39 0961 19 10 969

hathach, 2018-05-08:
can you post both serial output with debug level =1 of central and prph here. Please also describe how you  disconnect ble (there are many ways to do so). Just to make sure we are testing the same way.

antonioriccelli, 2018-05-08:
Yes
I have test with 2 identicall feathers boards and the example of arduino cental_bleuart and bleuart for both feathers
If they are connect and disconnect many times to BLE wireless connections, the BLE Central go to Crash and do not receive anymore….
After is necessary reboot this ultime board in cental_bleuart mode for re caming up.

hathach, 2018-05-08:
thanks for reporting the issue, we will try to test it out. How many times of connect + disconnect is large enough to reproduce the issue ? Just for us to know how many time we should power off and on the prph and expect the issue.

antonioriccelli, 2018-05-08:
sometimes enough twice and two or more times.
below the log. with debug 1 when the board is in block........

Bluefruit52 Central
--------------------------------------

[CFG   ] SoftDevice's RAM requires: 0x200021C8

BSP Library     : 0.8.3
Firmware        : S132 5.1.0, 5.1.0 dual banks
Serial No       : 70D3A40B5DAC200B

--------- SoftDevice Config ---------
Max UUID128     : 10
ATTR Table Size : 2048
Service Changed : 0
Central Connect Setting
  - Max MTU         : 23
  - Event Length    : 3
  - HVN Queue Size  : 1
  - WrCmd Queue Size: 1

--------- BLE Settings ---------
Name            : Bluefruit52 Central
Max Connections : Peripheral = 0, Central = 4 
Address         : F3:35:3B:8A:DC:E2 (Static)
TX Power        : 4 dBm
Conn Intervals  : min = 20.00 ms, max = 30.00 ms
Peripheral Paired Devices: 

Central Paired Devices: 

[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_CHAR_DISC_RSP : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_READ_RSP : Conn Handle = 0
200
Connected
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[BLE   ] BLE_GATTC_EVT_WRITE_CMD_TX_COMPLETE : Conn Handle = 0
[... 398 lines cut ...]
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535

wwwoholic, 2018-11-04:
I've noticed similar behavior in my tests. I also used "cental_bleuart" and "bleuart" examples, but replaced the sending code that takes manual Serial input with automatic sending taken from "throughput" example. The goal was to check maximum transfer speed in both directions simultaneously.

The code worked right away, however when I reboot Feather running central_bleuart, it cannot reconnect to the Feather running peripheral. Note, that if I run original code (without automatic sending) they reconnect just fine, regardless of which device is rebooted.

Here is printout on Central:
```
Bluefruit52 Central BLEUART Example
-----------------------------------
Scan callback
BLE UART service detected. Connecting ... Connect callback
0
Dicovering DIS ... Discovering BLE Uart Service ... Found NONE
0
Disconnected
Scan callback
Scan callback
Scan callback
Scan callback
Scan callback
Scan callback
```
Then "Scan callback" repeats forever.

I also tried to block sending from Central by adding boolean flag to connect_callback and disconnect_callback functions. Here is printout when I reboot Peripheral:
```
Received 2505
Sent 2500 bytes in 5.06 seconds.
Speed 0.49 KB/s.

Received 2505
Sent 2500 bytes in 5.18 seconds.
Speed 0.48 KB/s.

Received 2500
Sent 19  conn 1
Received 2465
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
Sent 19  conn 1
0
Disconnected
Sent 19  conn 0
Scan callback
BLE UART service detected. Connecting ... Connect callback
0
Dicovering DIS ... Found it
Manufacturer: Adafruit Industries
Model: Bluefruit Feather52

Discovering BLE Uart Service ... Found it
Enable TXD's notify
Ready to receive from peripheral
Sent 2500 bytes in 5.01 seconds.
Speed 0.50 KB/s.

Received 2490
Sent 2500 bytes in 5.18 seconds.
Speed 0.48 KB/s.

Received 2495
Sent 2500 bytes in 5.01 seconds.
Speed 0.50 KB/s.
```
Note "Sent 19 conn 1". The 19 is a direct result of the problem I mentioned in https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/192. Instead of returning 0 as number of bytes sent the write() function returns error code NRF_ERROR_RESOURCES.
The "conn 1" is a printout of "connected" boolean. It stays true until disconnected callback comes in, which happens long after write() function first returns error code.

wwwoholic, 2018-11-18:
Still occurring with 0.9.0
I noticed that it is easier to trigger this error if nodes are actively exchanging data (100~300 Bps) and one of the nodes is simply rebooted. If they just sit there connected but not exchanging data they almost always manage to reconnect.

hathach, 2019-01-11:
OK, I did reproduce the issue with central_bleuart and throughput prph. Press reset on central board while in the middle of sending data. Notice the busy error when discovering, maybe state is not cleaned up in prph.

My log for ref

```
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
BLE UART service detected. Connecting ... [BLE   ] BLE_GAP_EVT_CONNECTED : Conn Handle = 0
[GAP   ] Conn Interval= 20.000000
Connected
Dicovering Device Information ... [DISC  ] [SVC] Handle start = 1
[BLE   ] BLE_GATTC_EVT_PRIM_SRVC_DISC_RSP : Conn Handle = 0
[DISC  ] [SVC] Service Count: 1
[DISC  ] [SVC] Found 0x180A, Handle start = 22, end = 32
-----------------
Found it
[DISC  ] [CHR] Handle start = 22, end = 32
[DISC  ] [CHR] Handle start = 22, end = 32

Dicovering Battery ... [DISC  ] [SVC] Handle start = 1
bool BLEDiscovery::_discoverService(uint16_t, BLEClientService&, uint16_t): 79: verify failed, error = NRF_ERROR_BUSY
Discovering BLE Uart Service ... [DISC  ] [SVC] Handle start = 1
bool BLEDiscovery::_discoverService(uint16_t, BLEClientService&, uint16_t): 79: verify failed, error = NRF_ERROR_BUSY
Found NONE
[BLE   ] BLE_GAP_EVT_DISCONNECTED : Conn Handle = 0
Disconnected
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
[BLE   ] BLE_GAP_EVT_ADV_REPORT : Conn Handle = 65535
```

hathach, 2019-01-11:
It seems to be prph sketch issue with throughput, the while loop() wait forever and doesn't seem to handle ble event anymore. @wwwoholic can you pull and test again with update throuhgput example. 

@antonioriccelli I have tried to use central_bleuart and bleuart connected to each other then press reset button on central & prph in random order but it could re-connect just fine. Could you elaborate in detail how you reproduce the issue. Is there any data transmission between the two. Please also pull update before testing again.

hathach, 2019-02-27:
closed due to lack of user feedback

closed 2019-02-27 by hathach