adafruit/Adafruit_nRF52_Arduino #152: Does nRF52 pair with Windows 10?

kheek, 2018-05-11:
Has anyone pairing with Windows 10 (e.g. with HID example) up&running?

jpconstantineau, 2018-05-11:
Yes it works.  See [here](https://www.reddit.com/r/MechanicalKeyboards/comments/86asf6/curves_my_bluetooth_split/) and [here](https://gist.github.com/wez/b30683a4dfa329b86b9e0a2811a8c593).  Both use the Adafruit NRF52 and the libraries.

I am developing my [split keyboard](https://www.reddit.com/r/MechanicalKeyboards/comments/8i2twe/ergotravel_2_bluemicro_wireless_split_keyboard/) with Windows 10 and have used the libraries to build [this basic KB firmware](https://github.com/jpconstantineau/BlueMicro_BLE).  Had to use a third party NRF52 module because I wanted to make it fit into the Arduino Pro Micro pinout that many custom DIY keyboards use.

closed 2018-05-11 by hathach

jpconstantineau, 2018-05-11:
I occasionally get this error.  I disconnect, then reconnect and it's fine.

hathach, 2018-05-12:
@kheek  there are a few examples for hid, you can try it out too see if it work for you.

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard/hid_keyboard.ino