adafruit/Adafruit_nRF52_Arduino #153: add getRssi() and Rssi related callback

hathach, 2018-05-12:


baramuse, 2018-09-05:
Totally in for this one !
Still a bit new to the BT stack, will it give us the possibility to "measure" the distance between the devices ? 
Is that a big job ?

baramuse, 2018-09-05:
I actually started digging in the code and found the `sd_ble_gap_rssi_start` which I'm calling.
Now I've read the eventHandler code in BLEGap.cpp and it does not handle the `BLE_GAP_EVT_RSSI_CHANGED` event.
Is there a way to have an (sub?) eventHandler (like `void BLEGap::_eventHandler(ble_evt_t* evt)
` in the sketch so I can catch those events ?

hathach, 2018-09-05:
you must call it within connection callback, I guess it must be called before other ble arrived. Nordic docs doesn't mention this, I have been confusing like you before.

baramuse, 2018-09-05:
Thanks for the tip, i'll start when connected then.
Any tips for hooking an event listener instead of constantly reading the last sample value ?

hathach, 2018-09-05:
no we current doesn't have something like setBleEventCallback(), but may add it later on for user to work on unimplemented features. Much nicer than modifying the bsp code, which is a horrible lengthy code.

baramuse, 2018-09-05:
Ok at least i know :)
I'll start by reading frequently the rssi value, then I'll see if I implement the event callback !

Thanks again !

(btw is that a good way to detect the proximity of an object ? I want to set leds depending on the proximity of a remote - an iPhone in this case)

hathach, 2018-09-05:
proximity requires some processing filter and math calculation stuff, there is an example written by Kevin in the example projects. He has more exp than me in this field. You may want to check that out.