adafruit/Adafruit_nRF52_Arduino #160: Check Servo PWM frequency

ladyada, 2018-06-22:
hiya @hathach can you check the PWM frequency of the Servos? accordng to this person they are 640Hz instead of 50Hz?
https://forums.adafruit.com/viewtopic.php?f=57&t=137278

see also:
https://github.com/adafruit/Adafruit_nRF52_Arduino/issues/68

hathach, 2018-06-22:
Note to my self, check out #134

ladyada, 2018-07-01:
@hathach hi thach - do you mind looking at this bug this week? we had a few questions about it. let me know if you need help setting up a test!

hathach, 2018-07-02:
Ah sorry, I am a bit hooked up with usb. Will work on to resolve this.

ladyada, 2018-07-02:
no worries, i know USB is very consuming :) perhaps in the next week, i think its just a divider somewhere for a timer.

hathach, 2018-07-02:
thanks, I will update when i fixed this, hopefully still in this week.

hathach, 2018-07-12:
should be fixed with 456cb4a628a71c2427bad076e36daf974642d1de

closed 2018-07-12 by hathach