adafruit/Adafruit_nRF52_Arduino #161: RTC DateTime Functionality

tplooker, 2018-06-25:
Hi,

Is there any support for interfacing with the RTC to obtain datetime?

Thanks
Tobias

pcaddict, 2018-06-28:
The RTC peripheral in the nrf52 is a counter, not a clock.

tplooker, 2018-06-28:
HI @pcaddict Im aware of what the RTC I was wondering if anyone had written an implementation similar to this one (https://github.com/NordicPlayground/nrf5-calendar-example), which allows you to use the RTC to produce the current datetime on the device.

From what I understand, because the RTOS implementation relies on RTC1 to generate a low level tick interrupt for the RTOS scheduler, the above implementation wont work, as it also relies on a compare value interrupt from the same RTC. 

I have a working implementation using the supplied RTOS utility "SoftwareTimer" as the tick generator for the clock, however I dont think this is the best implementation and dont believe it will work in some low power modes?

hathach, 2018-07-01:
I think there is already a library for standard Arduino that handle RTC, you only need to feed it with rtos tick. I don't remember the name though.

tplooker, 2018-07-09:
@hathach Any chance you could point me in the rough direction as to where you found it, as thats exactly what I am after, I just don't want to butcher adafruits RTOS implementation by putting this calendar integration in the wrong place

hathach, 2018-07-12:
@tplooker I dont remember, it is only software rtc, you basically set the epoch, feed it with tick from e.g software timer, it will keeps track of the time/date etc ... I used one quite sometime ago.

nosv-dr, 2021-09-14:
I've used RTCLib for that https://github.com/nosv-dr/Adafruit_nRF52_Arduino/tree/master/libraries/RTClib. It hasn't been yet tested for overflow rollover though, so be careful.