adafruit/Adafruit_nRF52_Arduino #162: OTA DFU Key

lefedor, 2018-06-27:
To send firmware fro DFU manual tells I need sign it with very the same key was generated building bootloader. Can I obtain those keys (private.key at least)?
Or will bootloader accept just any DFU file generated in arduino sketch temp folder and I'm over-complicating things? Can You briefly guide steps for OTA with feather bootloader? (5.1.0 dual, s132).
Thanks for any information on the topick.

hathach, 2018-06-27:
We didn't use the secure bootloader, so no keys or whatsoever is required. We will make a tutorial for OTA later on.

closed 2018-06-27 by hathach

lefedor, 2018-06-27:
Thank You for the fast response. Feather support is really awesome.
Sadly I haven't knew there is forum option, will check it.