adafruit/Adafruit_nRF52_Arduino #163: consumer key with modifiers

squareup, 2018-07-10:
I'd like to be able to send volume up and down keys (and it was pretty easy).  I use
blehid.consumerKeyPress(HID_USAGE_CONSUMER_VOLUME_INCREMENT);

I'd like to be able to send shift + alt (or option) + volume up/down (this gives finer volume control on a Mac).

hathach, 2018-07-12:
hmm, interesting question. keyboard and consumer using different report ID. I will check to see if we could send 2 reports at once. Meanwhile, can you try to send one after another with

keyboardReport( shift | atl, 0 );
consumerKeyPress();

hathach, 2018-07-12:
PS: It seems that we couldn't send 2 reports in one data transaction. you should send them as quick as possible as above post to see if that works.

squareup, 2018-07-12:
I tried it and got an error when building it.  Error report follows:

bk_018:185: error: call of overloaded 'keyboardReport(int, int)' is ambiguous
     blehid.keyboardReport(KEYBOARD_MODIFIER_LEFTSHIFT | KEYBOARD_MODIFIER_LEFTALT, 0);
                                                                                     ^
In file included from /Users/kensale/Library/Arduino15/packages/adafruit/hardware/nrf52/0.8.3/libraries/Bluefruit52Lib/src/bluefruit.h:81:0,
                 from /Users/kensale/Documents/Arduino/bk_018/bk_018.ino:20:
/Users/kensale/Library/Arduino15/packages/adafruit/hardware/nrf52/0.8.3/libraries/Bluefruit52Lib/src/services/BLEHidAdafruit.h:58:10: note: candidate: bool BLEHidAdafruit::keyboardReport(uint8_t, uint8_t*)
     bool keyboardReport(uint8_t modifier, uint8_t keycode[6]);
          ^
/Users/kensale/Library/Arduino15/packages/adafruit/hardware/nrf52/0.8.3/libraries/Bluefruit52Lib/src/services/BLEHidAdafruit.h:59:10: note: candidate: bool BLEHidAdafruit::keyboardReport(uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t, uint8_t)
     bool keyboardReport(uint8_t modifier, uint8_t keycode0, uint8_t keycode1=0, uint8_t keycode2=0, uint8_t keycode3=0, uint8_t keycode4=0, uint8_t keycode5=0);
          ^
exit status 1
call of overloaded 'keyboardReport(int, int)' is ambiguous

hathach, 2018-07-12:
ah sorry, that is an ambiguous call, please try

blehid.keyboardReport(KEYBOARD_MODIFIER_LEFTSHIFT | KEYBOARD_MODIFIER_LEFTALT, 0, 0);

squareup, 2018-07-12:
It works :-)   
Thanks so much, you're a genius!

hathach, 2018-07-12:
thanks, phew, I am a bit worried that it may not work. I will close this issue now.

closed 2018-07-12 by hathach