adafruit/Adafruit_nRF52_Arduino #164: wiring_analog_nRF52.c unuseful code

a5021, 2018-07-17:
Code at line 245 does nothing

hathach, 2018-07-17:
```
NRF_SAADC->CH[0].PSELN = pin;
```
Is this one ? I am not really sure on this, we based our core & std peripheral on this repo https://github.com/sandeepmistry/arduino-nRF5

a5021, 2018-07-18:
Yes. It is the code I've wrote about.  "nRF52832 Product Specification v1.4" states it has no effect to init PSLEN without differential mode selected. Please see page 359, Also it is a nonsense to connect both differential ADC inputs to a single pin.

hathach, 2019-01-11:
Ok, I checked the specs, but I don't really understand how the Analog input SAADC works. I will play it safe (or coward) here and put this issue on hold until it become an issue or more people come to verify this. Thank you very much for the issue.

closed 2019-07-09 by hathach