adafruit/Adafruit_nRF52_Arduino #166: How to receive signals (Caps lock, Num lock) from PC side - HID keyboard

ginoji5, 2018-07-26:
For making wireless HID keyboard modules (with nRF52 feather), I want to know the method receive signals from PC side. (Num lock, Caps lock, etc.)
It is possible to input arbitrary keys at the present time, but we also want to get output from PC side as well.

hathach, 2018-07-26:
Thank for raising the issue, it is output report. Seems like the BLEHidGeneric already support invoked callback when receiving an output from PC.
  
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/services/BLEHidGeneric.h#L114

Though it is not used by BLEHidAdafruit (with proper report id and callback for keyboard). The work should be minimal, we will try to fix it when we could take some rest from nrf52840 work.

ginoji5, 2018-07-26:
I see.
I will look BLEHidGeneric.h, and study output report.

hathach, 2018-08-01:
@ginoji5  I did a latest push to master that implement the keyboard led. Tested with my linux. Let's me know if it works for you (please also tell what is your OS as well).

hathach, 2018-08-01:
this is all you need to call
https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard/hid_keyboard.ino#L56

ginoji5, 2018-08-01:
Thank you for the great update!
I tested with my Win10. The result was good.
Next time, I will test it on Raspberry pi zero w.

hathach, 2018-08-02:
Great, Thanks for the update. I am glad it works for you. Some users actually have issue with windows and blehid. I will pull out a macos to test, if all major os works+rpi, then we could close this one.

hathach, 2018-08-15:
@ginoji5  any issues so far, if not we will close this one and include it in the next release

ginoji5, 2018-08-18:
On Raspberry pi , the test was good too. Thank you.

closed 2018-08-18 by ginoji5

AustinRhoads, 2022-12-13:
The solution is 404. Can anyone point me in the right irection?