adafruit/Adafruit_nRF52_Arduino #167: Turn Bluetooth on and off

jeremypoulter, 2018-07-26:
Related to #165 I am also working on Low power, is it possible to stop the Bluetooth once started?

hathach, 2018-07-27:
indeed, there should be a Bluefruit.stop() . We will include it in the next release. Thanks for posting the issue.

jeremypoulter, 2018-07-27:
If you can post some direction more than happy to do a PR

hathach, 2018-07-27:
It is essentially revert of the Bluefruit.begin()

https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/src/bluefruit.cpp#L235 

free up all resources task stack memory, delete task etc ... then call sd_softdevice_disable ( counter-part of sd_softdevice_enable )

jeremypoulter, 2018-07-27:
> Bluefruit52Lib/src/bluefruit.cpp#L235

I was thinking about that, but was thinking that is a bit drastic and was
hoping for more of a pause and turn off the radio rather than tear down the
full stack, Essentially this is for an off mode, I don't want to do the
full power off as I need the RTC to keep running, but do want to shut down
everything else.

hathach, 2018-07-27:
Ah I guess I got what you mean now, it is some sort of Bluefruit.pause() rather, I don't have any particular ideas now. But I think you could disable the BLE stack without disable the whole SoftDevice. Too bad, I am currently too busy with nrf52840, if you don't mind checking out the S132 specifications and docs, it may have something stated there. I will try to go back to 52832 a bit after getting 52840 a bit more stable.

pbecchi, 2018-07-30:
From the test I did for issue #165 ,power consumption depend mainly from advertising, if you stop advertising the uA should go back to ≤10 when you are in low power mode...... eg.delay()!

jappavoo, 2019-05-21:
Any updates on this?  For my use a full stop would be great.

hathach, 2019-05-21:
I have't got time to work on this

joshysav, 2020-01-14:
Sorry @hathach , any updates? :|

MuckYu, 2020-05-20:
Would also like to see a .pause function.

Right now I use .stop to stop the BLE for a short time before enabling it again.

glenne, 2021-01-28:
I need this as well.  @MuckYu , Where did you find a .stop() method?

MuckYu, 2021-01-30:
Sorry used this a long time ago and don't remember the exact details.
I believe in my case I tried to scan an RFID chip and send the code via
bluetooth. But somehow it started sending the code while it was still
scanning the RFID chip => weird chunks coming via Bluetooth.
So I had to STOP the bluetooth function for just a second and send the
whole RFID chip code once it reconnects.

Below are some snippets from the code I used.

"Bluefruit.Advertising.stop();"

It should basically stop the bluetooth function and if you have
auto-restart enabled it will reconnect.

void startAdv(void)
{
  // Advertising packet

Bluefruit.Advertising.addFlags(BLE_GAP_ADV_FLAGS_LE_ONLY_GENERAL_DISC_MODE);
  Bluefruit.Advertising.addTxPower();

  // Include bleuart 128-bit uuid
  Bluefruit.Advertising.addService(bleuart);

  // Secondary Scan Response packet (optional)
  // Since there is no room for 'Name' in Advertising packet
  Bluefruit.ScanResponse.addName();

  /* Start Advertising
   * - Enable auto advertising if disconnected
   * - Interval:  fast mode = 20 ms, slow mode = 152.5 ms
   * - Timeout for fast mode is 30 seconds
   * - Start(timeout) with timeout = 0 will advertise forever (until
connected)
   *
   * For recommended advertising interval
   * https://developer.apple.com/library/content/qa/qa1931/_index.html
   */
  Bluefruit.Advertising.restartOnDisconnect(true);
  Bluefruit.Advertising.setInterval(32, 244);    // in unit of 0.625 ms
  Bluefruit.Advertising.setFastTimeout(30);      // number of seconds in
fast mode
  Bluefruit.Advertising.start(0);                // 0 = Don't stop
advertising after n seconds

}

Hope this helps

ando600, 2022-03-27:
Hi, is there a way to stop the BLE service? I have a need similar to glenee. 

Is there a workaround that I can do to fix this? Even if it's not included in the library, I really need to put the device in low power mode, and Bluefruit.begin causes the device to exit low power mode immediately, which does not work for my needs. 

Thanks,
Andrew

mittlc, 2023-09-15:
Dear Adafruit team,
is there any news on this issue?

I am looking for a way to stop the Bluefruit device and start it back up (Bluefruit.begin()) later to try to work around compatibility issues with my gps module, which seem to be blocked as soon as i start using the BLE.

glenne, 2023-09-15:
I use the following before going to deep sleep.  Maybe it'll help you.  After deep sleep and waiting on a button press, it comes up like coming out of reset and reconfigures everything.

```
static void disconnectBle() {
  uint16_t connections = Bluefruit.connected();
  for (uint16_t conn = 0; conn < connections; conn++) {
    Bluefruit.disconnect(conn);
  }
}

void ble_sleep(void) {
  Bluefruit.Advertising.restartOnDisconnect(false);
  disconnectBle();
  Bluefruit.Advertising.stop();
}
```

After setting some lines hi-z and setting up a power-up interrupt I use
```
sd_power_system_off();
```

Maybe the ble_sleep() is sufficient for your needs followed by restarting advertising when you want to start again.

mittlc, 2023-09-18:
Thanks @glenne,
Disconnecting on the client Side was actually sufficient. The issue was related to timing and to many activities at once. As soon as i introduced some time for each Operation GPS and Bluetooth go along just fine.