adafruit/Adafruit_nRF52_Arduino #171: attachInterrupt vs. SoftDevice

Nenik, 2018-08-29:
I have run into an issue when using GPIO interrupts.
If I call attachInterrupt() before Bluefruit.begin(), the Bluefruit.begin() fails with NRF_ERROR_SDM_INCORRECT_INTERRUPT_CONFIGURATION, which I suspect might be related to the interrupt priority used by the attachInterrput() implementation.
Calling Bluefruit.begin() first, the interrupr callback works well.
This would really be worth documenting at least.
But should the change in the interrupt priority fix the issue, that would be even better.

hathach, 2018-08-29:
Ah yeah, it is mentioned multiple times in the forum. Bluefruit.begin() enable sd, which will affect the vector table in some ways. Some irq may not work if called before that. We haven't looked into the details to resolve this, maybe later when we got more free time.

Nenik, 2018-09-01:
Pull request: https://github.com/adafruit/Adafruit_nRF52_Arduino/pull/179

hathach, 2018-11-08:
thanks for PR, this should be fixed already.

closed 2018-11-08 by hathach