adafruit/Adafruit_nRF52_Arduino #174: Version 0.8.5 can't compile on windows

craiq, 2018-08-30:
Hey,

OS: Windows 10 Pro 64-Bit (10.0, Build 15063)
Arduino IDE 1.8.6

after updating to version 0.8.5 I get following compiler error:

```
[18560] Failed to execute script __main__

Traceback (most recent call last):

  File "nordicsemi\__main__.py", line 315, in <module>

  File "site-packages\click\core.py", line 722, in __call__

  File "site-packages\click\core.py", line 697, in main

  File "site-packages\click\core.py", line 1066, in invoke

  File "site-packages\click\core.py", line 1066, in invoke

  File "site-packages\click\core.py", line 895, in invoke

  File "site-packages\click\core.py", line 535, in invoke

  File "nordicsemi\__main__.py", line 240, in genpkg

  File "site-packages\click\utils.py", line 259, in echo

  File "site-packages\click\_winconsole.py", line 180, in write

  File "site-packages\click\_winconsole.py", line 164, in write

OSError: Windows error 1

exit status -1
Fehler beim Kompilieren für das Board Adafruit Bluefruit nRF52 Feather.
```

going back to 0.8.4 solves the problem.

Tested this on the bleuart example.

hathach, 2018-08-30:
can you post your full log of compilation.

hathach, 2018-08-30:
Ah sorry, i didn't state it clear, can you enabled the verbose mode for compilation in preference. It provide a bit more information. You dont have to paste the full one, only the last command that trigger the error. it is probable `adafruit-nrfutil .......`

[image]

craiq, 2018-08-30:
with this options activated there is no error anymore oO
deactivating it and it comes back

hathach, 2018-08-30:
@craiq did you installed the win10 version from windows app store, or just a normal exe file from Arduino website ?

craiq, 2018-08-30:
from the website

hathach, 2018-08-30:
Thanks for reporting, we will try to fix this asap.

clratzlaff, 2018-09-06:
I just installed Arduino IDE 1.8.6 onto my Windows 10 (10.0.17134 Build 17134) and had the exact same issue.  Enabling the verbose output solved the problem also.

ladyada, 2018-09-11:
@craiq @clratzlaff try replacing your adafruit-nrfutil exec in Adafruit_nRF52_Arduino\tools\adafruit-nrfutil\win32 with this one and try again?

https://s3.amazonaws.com/uploads.hipchat.com/95690/1622213/Fkh16cYONkP3owE/adafruit-nrfutil.exe

if that fixes it we'll bump and re-release

clratzlaff, 2018-09-11:
Thanks for the fix.  That fixed the problem.  I can now compile and upload without the verbose flags set.

ladyada, 2018-09-11:
yay thanks for checking :)

closed 2018-09-14 by hathach

hathach, 2018-09-15:
We just release 0.8.6 that should fix this issue along with other minors issue. Thanks for posting.

craiq, 2018-09-25:
back from holiday. thx its solved!

commit be4dca90 (2018-09-14): update adafruit-nrfutil.exe to post10 for windows. fix #174