adafruit/Adafruit_nRF52_Arduino #175: Segger Embbeded Studio for advance debugging

hathach, 2018-08-31:
SES is much better when coming to debugging, great for those that have jlink and better to develop our code.

jeremypoulter, 2018-08-31:
FYI I have done some work to get the Adafruit Core running under Platform IO which is also good for advanced debugging - https://github.com/jeremypoulter/platform-nordicnrf52/tree/adafruit_nrf52_feather_intergration

hathach, 2018-08-31:
Superb!! Thanks, I will be sure check that our for reference.

Nenik, 2018-09-01:
@jeremypoulter Can you expand on the PIO integration?
I have tried setting up, even with "platform = https://github.com/platformio/platform-nordicnrf52.git
", but if I try to set the board to adafruit_feather_nrf52, I get:
Error: Unknown board ID 'adafruit_feather_nrf52'
I know little about Atom/PIO internals, but the github version does have adafruit_feather_nrf52.json
 under boards....

Nenik, 2018-09-01:
OK, this seems to get me further, for reference:
platform = https://github.com/jeremypoulter/platform-nordicnrf52#adafruit_nrf52_feather_intergration
board = adafruit_feather_nrf52
framework = arduino

jeremypoulter, 2018-09-01:
Right that is correct, FYI I also updated the example project, https://github.com/jeremypoulter/platformio-examples/tree/develop/wiring-blink

romansoft, 2018-10-06:
@hathach, is there a SES project file you could share to get started with compiling and debugging in Segger Embedded Studio?

hathach, 2018-10-06:
@romansoft unfortunately, I haven't started to do anything for this issue yet.

jeremypoulter, 2019-03-09:
FYI PlatformIO now supports the Adafruit nRF boards, debugging works great ;-)

hathach, 2019-03-11:
@jeremypoulter Superb!!! One of the issue with segger set up is the project is library only, it need the user sketch to complete the binary. I will close it for now, it is probably easier for user to use PlatformIO or segger ozone to troubleshoot the issue.

closed 2019-03-11 by hathach